jfedor2/hid-remapper #2: Enable upper buttons on the Kensington Slimblade

lstubbig, 2022-03-11:
First of all, thanks for starting this! It looks like a very promising tool for remapping trackball functions with a hardware solution. My application is a Windows laptop where I'd like to avoid installing the Kensington driver and the KensingtonWorks software.

I have tested it with the Slimblade trackball, and the upper two buttons don't work at the moment. Unfortunately Kensington seems to use some non-standard implementation here. Do you think these buttons can be added? On Reddit you mentioned hid-recorder. Below is the log from pressing the bottom left, bottom right, top left, and top right buttons, in this order (recorded on Linux Mint 20.3). Let me know how else I can help.

```
# Kensington Slimblade Trackball
# 0x05, 0x01,                    // Usage Page (Generic Desktop)        0
# 0x09, 0x02,                    // Usage (Mouse)                       2
# 0xa1, 0x01,                    // Collection (Application)            4
# 0x09, 0x01,                    //  Usage (Pointer)                    6
# 0xa1, 0x00,                    //  Collection (Physical)              8
# 0x05, 0x09,                    //   Usage Page (Button)               10
# 0x19, 0x01,                    //   Usage Minimum (1)                 12
# 0x29, 0x02,                    //   Usage Maximum (2)                 14
# 0x15, 0x00,                    //   Logical Minimum (0)               16
# 0x25, 0x01,                    //   Logical Maximum (1)               18
# 0x95, 0x02,                    //   Report Count (2)                  20
# 0x75, 0x01,                    //   Report Size (1)                   22
# 0x81, 0x02,                    //   Input (Data,Var,Abs)              24
# 0x95, 0x01,                    //   Report Count (1)                  26
# 0x75, 0x06,                    //   Report Size (6)                   28
# 0x81, 0x03,                    //   Input (Cnst,Var,Abs)              30
# 0x05, 0x01,                    //   Usage Page (Generic Desktop)      32
# 0x09, 0x30,                    //   Usage (X)                         34
# 0x09, 0x31,                    //   Usage (Y)                         36
# 0x09, 0x38,                    //   Usage (Wheel)                     38
# 0x15, 0x81,                    //   Logical Minimum (-127)            40
# 0x25, 0x7f,                    //   Logical Maximum (127)             42
# 0x75, 0x08,                    //   Report Size (8)                   44
# 0x95, 0x03,                    //   Report Count (3)                  46
# 0x81, 0x06,                    //   Input (Data,Var,Rel)              48
# 0x06, 0x00, 0xff,              //   Usage Page (Vendor Defined Page 1) 50
# 0x19, 0x01,                    //   Usage Minimum (1)                 53
# 0x29, 0x02,                    //   Usage Maximum (2)                 55
# 0x15, 0x00,                    //   Logical Minimum (0)               57
# 0x25, 0x01,                    //   Logical Maximum (1)               59
# 0x95, 0x02,                    //   Report Count (2)                  61
# 0x75, 0x01,                    //   Report Size (1)                   63
# 0x81, 0x02,                    //   Input (Data,Var,Abs)              65
# 0x95, 0x01,                    //   Report Count (1)                  67
# 0x75, 0x06,                    //   Report Size (6)                   69
# 0x81, 0x03,                    //   Input (Cnst,Var,Abs)              71
# 0xc0,                          //  End Collection                     73
# 0xc0,                          // End Collection                      74
# 
R: 75 05 01 09 02 a1 01 09 01 a1 00 05 09 19 01 29 02 15 00 25 01 95 02 75 01 81 02 95 01 75 06 81 03 05 01 09 30 09 31 09 38 15 81 25 7f 75 08 95 03 81 06 06 00 ff 19 01 29 02 15 00 25 01 95 02 75 01 81 02 95 01 75 06 81 03 c0 c0
N: Kensington Slimblade Trackball
I: 3 047d 2041
#  Button: 1  0 | # | X:    0 | Y:    0 | Wheel:    0 | Vendor Usage 1: 0 | Vendor Usage 2: 0 | # 
E: 000000.000000 5 01 00 00 00 00
#  Button: 0  0 | # | X:    0 | Y:    0 | Wheel:    0 | Vendor Usage 1: 0 | Vendor Usage 2: 0 | # 
E: 000000.192059 5 00 00 00 00 00

#  Button: 0  1 | # | X:    0 | Y:    0 | Wheel:    0 | Vendor Usage 1: 0 | Vendor Usage 2: 0 | # 
E: 000001.439958 5 02 00 00 00 00
#  Button: 0  0 | # | X:    0 | Y:    0 | Wheel:    0 | Vendor Usage 1: 0 | Vendor Usage 2: 0 | # 
E: 000001.727981 5 00 00 00 00 00

#  Button: 0  0 | # | X:    0 | Y:    0 | Wheel:    0 | Vendor Usage 1: 1 | Vendor Usage 2: 0 | # 
E: 000005.543959 5 00 00 00 00 01
#  Button: 0  0 | # | X:    0 | Y:    0 | Wheel:    0 | Vendor Usage 1: 0 | Vendor Usage 2: 0 | # 
E: 000005.815943 5 00 00 00 00 00

#  Button: 0  0 | # | X:    0 | Y:    0 | Wheel:    0 | Vendor Usage 1: 0 | Vendor Usage 2: 1 | # 
E: 000007.088063 5 00 00 00 00 02
#  Button: 0  0 | # | X:    0 | Y:    0 | Wheel:    0 | Vendor Usage 1: 0 | Vendor Usage 2: 0 | # 
E: 000007.407931 5 00 00 00 00 00
```

jfedor2, 2022-03-11:
Hi, yeah, this is precisely the kind of info I need to know what's going on. It is as you say, Kensington is using a non-standard (or at least vendor-specific) way of reporting those two buttons.

I'm currently working on a more generic way of mapping inputs that would make it possible to handle this situation as well, but it's probably going to be some time before that's ready.

In the mean time I can make a special version of the current firmware with the Slimblade's format hardwired.

jfedor2, 2022-03-11:
I haven't tested it at all, but try this. Buttons should work as 3 and 4 and you should be able to remap them.

[remapper-slimblade.zip](https://github.com/jfedor2/hid-remapper/files/8235412/remapper-slimblade.zip)

lstubbig, 2022-03-11:
I did a short test and it seems to work great! Thanks, I will follow this project with a lot of interest.

Anodynous, 2022-03-22:
I can also confirm that the remapper-slimblade build works great for remapping button 3 and 4 on the device. 

Would you be able to detail the changes required to the code to add this support? I am compiling from source to incorporate some other minor modifications which is why I am interested.

jfedor2, 2022-03-22:
I just put in a fake HID report descriptor with the vendor-specific usages changed to standard button ones and made it parse that instead of the descriptor received from the device. Here's the diff.

```
diff --git a/firmware/src/descriptor.cc b/firmware/src/descriptor.cc
index 5f0c454..4fa8666 100644
--- a/firmware/src/descriptor.cc
+++ b/firmware/src/descriptor.cc
@@ -24,6 +24,51 @@ axis_def_t axes[4];  // X, Y, V, H

 bool has_report_id;

+// Kensington Slimblade
+uint8_t report_descriptor[] = {
+ 0x05, 0x01,                    // Usage Page (Generic Desktop)        0
+ 0x09, 0x02,                    // Usage (Mouse)                       2
+ 0xa1, 0x01,                    // Collection (Application)            4
+ 0x09, 0x01,                    //  Usage (Pointer)                    6
+ 0xa1, 0x00,                    //  Collection (Physical)              8
+ 0x05, 0x09,                    //   Usage Page (Button)               10
+ 0x19, 0x01,                    //   Usage Minimum (1)                 12
+ 0x29, 0x02,                    //   Usage Maximum (2)                 14
+ 0x15, 0x00,                    //   Logical Minimum (0)               16
+ 0x25, 0x01,                    //   Logical Maximum (1)               18
+ 0x95, 0x02,                    //   Report Count (2)                  20
+ 0x75, 0x01,                    //   Report Size (1)                   22
+ 0x81, 0x02,                    //   Input (Data,Var,Abs)              24
+ 0x95, 0x01,                    //   Report Count (1)                  26
+ 0x75, 0x06,                    //   Report Size (6)                   28
+ 0x81, 0x03,                    //   Input (Cnst,Var,Abs)              30
+ 0x05, 0x01,                    //   Usage Page (Generic Desktop)      32
+ 0x09, 0x30,                    //   Usage (X)                         34
+ 0x09, 0x31,                    //   Usage (Y)                         36
+ 0x09, 0x38,                    //   Usage (Wheel)                     38
+ 0x15, 0x81,                    //   Logical Minimum (-127)            40
+ 0x25, 0x7f,                    //   Logical Maximum (127)             42
+ 0x75, 0x08,                    //   Report Size (8)                   44
+ 0x95, 0x03,                    //   Report Count (3)                  46
+ 0x81, 0x06,                    //   Input (Data,Var,Rel)              48
+// 0x06, 0x00, 0xff,              //   Usage Page (Vendor Defined Page 1) 50
+ 0x05, 0x09,                    //   Usage Page (Button)               10
+// 0x19, 0x01,                    //   Usage Minimum (1)                 53
+// 0x29, 0x02,                    //   Usage Maximum (2)                 55
+ 0x19, 0x03,                    //   Usage Minimum (3)                 53
+ 0x29, 0x04,                    //   Usage Maximum (4)                 55
+ 0x15, 0x00,                    //   Logical Minimum (0)               57
+ 0x25, 0x01,                    //   Logical Maximum (1)               59
+ 0x95, 0x02,                    //   Report Count (2)                  61
+ 0x75, 0x01,                    //   Report Size (1)                   63
+ 0x81, 0x02,                    //   Input (Data,Var,Abs)              65
+ 0x95, 0x01,                    //   Report Count (1)                  67
+ 0x75, 0x06,                    //   Report Size (6)                   69
+ 0x81, 0x03,                    //   Input (Cnst,Var,Abs)              71
+ 0xc0,                          //  End Collection                     73
+ 0xc0,                          // End Collection                      74
+};
+
 void mark_button(uint32_t usage, uint8_t report_id, uint16_t bitpos) {
     printf("mark_button(%0lx, %0hhx, %0hx)\n", usage, report_id, bitpos);
     if (usage >> 16 == 0x09) {
@@ -61,8 +106,9 @@ void mark_axis(uint32_t usage, uint8_t report_id, uint16_t bitpos, uint16_t size
     }
 }

-void parse_descriptor(volatile uint8_t* report_descriptor, int len) {
+void parse_descriptor() {
     int idx = 0;
+    int len = sizeof(report_descriptor);

     uint8_t report_id = 0;
     std::unordered_map<uint8_t, uint16_t> bitpos;  // report_id -> bitpos
diff --git a/firmware/src/descriptor.h b/firmware/src/descriptor.h
index 264bd07..6ca4604 100644
--- a/firmware/src/descriptor.h
+++ b/firmware/src/descriptor.h
@@ -28,7 +28,7 @@ extern bool has_report_id;
 extern "C" {
 #endif

-void parse_descriptor(volatile uint8_t* report_descriptor, int len);
+void parse_descriptor();
 void clear_descriptor_data();

 #ifdef __cplusplus
diff --git a/firmware/src/pio_usb/pio_usb.c b/firmware/src/pio_usb/pio_usb.c
index e08180d..2648ff6 100644
--- a/firmware/src/pio_usb/pio_usb.c
+++ b/firmware/src/pio_usb/pio_usb.c
@@ -1251,7 +1251,7 @@ static int enumerate_device(usb_device_t *device, uint8_t address) {
         }
         printf("\n");
         stdio_flush();
-        parse_descriptor(device->control_pipe.rx_buffer, desc_len);
+        parse_descriptor();

       } break;
       default:
```

Anodynous, 2022-03-23:
Thank you, that did the trick!
I also want to echo what lstubbig said about this being a really promising tool and interesting project. Couldn't be happier to be rid of the kensingtonworks software.

jfedor2, 2022-04-11:
This should now be addressed in two different ways.

First, the current version of the firmware and config utility allows remapping arbitrary inputs, including vendor-specific ones. To make this easier, the config tool displays them below all the standard inputs as hex codes (to make the UI part work, you have to click "Open device" while your mouse/trackball/whatever is connected to the remapper, then it can fetch all the inputs declared by the device and add them to the list).

Second, I've added special handling for the Kensington Slimblade trackball, similar to what was done in the firmware posted above, making the buttons available as 3 and 4 ("middle" and "back"), for convenience.

Anodynous, 2022-04-17:
Just checked out the new version. Nice improvements and the top slimblade keys indeed work natively now.

jfedor2, 2022-06-06:
Fun fact, I heard that Kensington has done a silent revision of the Slimblade, the new version uses standard codes for the top two buttons.

closed 2022-06-06 by jfedor2

b-, 2022-06-06:
> Fun fact, I heard that Kensington has done a silent revision of the Slimblade, the new version uses standard codes for the top two buttons.

May I ask where you heard this? I can't find anything about it online.