jfedor2/hid-remapper #3: Overclocking MLT04 mice (IMO 1.1a specifically), mouse won't respond?

yourrandomnobody, 2022-04-19:
Hello, these are the steps I've done:
-Got microUSB to USB cable from my motorola charger
-Held the BOOTSEL button (while having microUSB put in) and inserted the USB part in my PC
-Linux/Windows found the RPI Pico, opened the folder up and inserted the remapper.uf2 file inside
-Folder closed, all things are still plugged in
-Opened up the website and connected the mouse to the soldered USB hub
-Mouse has LED's turned on but it doesn't recognize any input?
I've tried mapping cursor X and Y, etc. to the same outputs but to no avail. I feel like I'm either connecting the mouse early/late or not using the website right, any help?

https://youtu.be/nxhV84Gcb0g

https://cdn.discordapp.com/attachments/263456086001057804/966016984535928892/20220419_184516.jpg

jfedor2, 2022-04-19:
Hi,

In theory it should work without using the website at all (it's not very useful that way, but it should at least move the cursor).

Do you have some other USB mouse (or keyboard) to try and check if it's a problem with the Pico setup or this particular mouse?

yourrandomnobody, 2022-04-19:
The mouse by itself works fine, when directly plugging into the USB port of my motherboard (I am using the USB2.0 port btw)
Tried another mouse, problem still persists.
Is it possible that the soldering is bad? Or perhaps the female USB plug is faulty?

jfedor2, 2022-04-19:
Yeah, I meant that perhaps the remapper doesn't like this particular mouse for some reason, not that it doesn't work otherwise. I've had it happen with one mouse in the past.

If no other device works when plugged into the remapper then I would definitely check the wiring.

michelwoo, 2022-04-23:
To add to this. I also have issues with the io1.1a not moving or clicking at all. Another mouse that also doesnt move is the finalmouse ultralight capetown. I've tried a different mouse (Vaxee NP01) and that works. @jfedor2 Is it possible to adjust the remapper to work with other mice?

jfedor2, 2022-04-23:
As mentioned above, it looks like the library that I'm using for the device side (the one which you connect your mouse to) doesn't like certain devices, I'm not sure why. The issue has been reported to the author: https://github.com/sekigon-gonnoc/Pico-PIO-USB/issues/18

I can't speculate on what the chances are for a fix.

On my side I'm considering making a version of the remapper that wouldn't use that particular library and instead would use two Pico boards, one for the host and one for the device USB interface (with the two sides communicating over serial or whatever). The idea being that the native hardware USB of the Pico should hopefully work with more (all) devices as it's more mature than the bit-banging library.

michelwoo, 2022-04-23:
Ahh I see. Yeah a "double" pico board would be nice if it would work with all mice.

jfedor2, 2022-04-24:
@yourrandomnobody @michelwoo There might still be hope for the current hardware. If you have a moment, please try this firmware, in my tests it worked with at least one mouse that didn't work with the regular firmware.

[remapper-experimental-2022-04-24.zip](https://github.com/jfedor2/hid-remapper/files/8550493/remapper-experimental-2022-04-24.zip)

There are some caveats, in my tests it sometimes stopped working randomly, so try connecting and disconnecting a few times if possible.

Also the overclocking setting currently doesn't work - it will work as if set to 1000 Hz regardless of which option you choose.

But it might provide some guidance on whether this path is worth pursuing.

yourrandomnobody, 2022-04-25:
> 

This one made my WMO and IMO 1.1a work! Cursor, left and right mouse button works. The scroll wheel doesn't work however.

jfedor2, 2022-04-25:
@yourrandomnobody Yeah, the scroll wheel is a silly mistake, changing the library inadvertently made the mouse work in boot protocol mode. That should be an easy fix. But overall it's a good result, it means that it works on the hardware level.

I think we can followup on the polling rate stability once I manage to make the rate configurable again in this version. Stay tuned.

yourrandomnobody, 2022-04-26:
@jfedor2 Great to read! Also, the polling rate stability was an issue in the original .uf2 file as well, just wanted to clarify.
It's not something from this experimental build.
Eagerly waiting for the new version to test!

michelwoo, 2022-04-27:
Awesome to hear @jfedor2 , looking forward to trying the next verison

shriramhr, 2022-05-11:
> @yourrandomnobody Yeah, the scroll wheel is a silly mistake, changing the library inadvertently made the mouse work in boot protocol mode. That should be an easy fix. But overall it's a good result, [...]

Hi,
First of all, thanks a ton for making this 🙏
Been trying to figure out a way to do this myself for a while, never got to it though thinking its way more complicated than it looked. Obviously I was wrong.

I newer experimental firmware solves the issues for most mice. As it has been mentioned, only the scroll is an issue. Just wondering what steps were taken to generate/compile the uf2 file? Is the same library available for other rbp models, especially ones which can be configured as high speed devices? If so then we can oc at higher clock speeds I believe.

jfedor2, 2022-05-13:
> I newer experimental firmware solves the issues for most mice. As it has been mentioned, only the scroll is an issue. Just wondering what steps were taken to generate/compile the uf2 file?

@shriramhr

Scroll is an easy fix like I mentioned, I will make a new version with that and hopefully make the polling rate override configurable again, I'm just waiting for the merges in the tinyusb and pio-usb repositories to happen so that we're tracking a somewhat stable target (though some changes on top of that will probably still be needed for the overclocking).

If you want to compile this yourself, the steps are pretty much the same as for the pico-sdk examples. Something along the lines of:

```
git clone https://github.com/raspberrypi/pico-sdk.git
cd pico-sdk && git submodule update --init
export PICO_SDK_PATH=`pwd`
cd ..
git clone https://github.com/jfedor2/hid-remapper.git
cd hid-remapper
mkdir build
cd build
cmake ..
make
```

(Note that the "experimental" changes aren't pushed to the repository yet, so this would get you the normal, "stable" firmware currently.)

> Is the same library available for other rbp models, especially ones which can be configured as high speed devices? If so then we can oc at higher clock speeds I believe.

If you're referring to the "big" (non-Pico) Raspberry Pis that run Linux then no, the library won't run on them. Though the general idea should be doable on them as they can act as both USB hosts and devices AFAIK.

shriramhr, 2022-05-14:
> @shriramhr
>
> Scroll is an easy fix like I mentioned, I will make a new version with that and hopefully make the polling rate override configurable again, I'm just waiting for the merges in the [...]

Thanks for the reply. I wanted to check if i can fiddle around with the code. Right now more than any other mouse I wanted to get the dell ms116 working with 1000hz. The experimental firmware works for it but its glitchy. Like holding left/right click doesnt work, it just does multiple clicks and in general it stutters a bit. I think it can be fixed in code. The intelli mice work flawlessly though.

jfedor2, 2022-06-13:
Hi guys,

I made a new version that might be of interest to people who had issues with compatibility previously. Since it uses the RP2040's native USB interface for both sides (host and device), it should hopefully work with more devices than the previous bit-banged implementation.

To achieve this the new version uses two Picos that talk to each other over serial. I haven't gotten around to properly documenting it yet, but the connection diagram below should have all the information needed if anyone's brave enough to try it:

[image]

Flash the "A" side with `remapper_dual_a.uf2` and the "B" side with `remapper_dual_b.uf2`. Don't connect both sides to a host at the same time. Use the same website for configuration as with the previous version.

As a bonus it should now also support multiple devices at the same time via a USB hub (though with certain limitations that I hope to address in the future).

jfedor2, 2022-06-13:
> Just a quick question - will this double pico setup cause any sort of extra input delay? Will it be like around 200us ish or so or like 1ms since both will be running at 1KHz?

In my tests it adds ~0.9ms latency on average. The single Pico version is essentially the same.

[image]

(Using this setup: https://inputlag.science/controller/methodology )

shriramhr, 2022-06-14:
> In my tests it adds ~0.9ms latency on average. The single Pico version is essentially the same.
>
> [image]
>
> (Using this setup: https://inputlag.science/controller/methodology )

Thanks a ton for the extra effort. I'll make this and let you know how it goes

shriram-es, 2022-06-14:
Not sure if it's some loose contact in my otg cable or the jumper wires, it doesn't seem to work right now. Probably some loose contact, ordered some new wires and an otg cable.

shriram-es, 2022-06-15:
> Hi guys,
>
> I made a new version that might be of interest to people who had issues with compatibility previously. Since it uses the RP2040's native USB interface for both sides (host and device), it [...]

DUUUUUDE! I got it to work like a charm! Basically old wires were useless... now I've tried all MLT04 mice + old Dell MS116 mouse.... works flawlessly

jfedor2, 2022-06-15:
> DUUUUUDE! I got it to work like a charm! Basically old wires were useless... now I've tried all MLT04 mice + old Dell MS116 mouse.... works flawlessly

Glad to hear that. Thank you for testing!

I will add some docs and hopefully a case for the dual Pico version in the future.

urfdvw, 2022-06-21:
Just a comment.

Are you interested in documenting the signal from board B?
This can be a good replacement for [USB Host Co-Processor](https://github.com/gdsports/usbhostcopro),
which is not active any more, and sadly not working for many of my devices.

I am only familiar with Arduino and CircuitPython,
and that is why I am not very ready to fork and contribute to this repo directly.
If signal from board B is understood,
that opens possibility for me to make a board A' that runs CircuitPython.

This is just my thoughts,
by no means to be any push or priority.

jfedor2, 2022-06-25:
> Are you interested in documenting the signal from board B?
> This can be a good replacement for [USB Host Co-Processor](https://github.com/gdsports/usbhostcopro),
> which is not active any more, and sadly not working for many of my devices.

Yeah, I still need to document the dual Pico version properly.

The A-B protocol is very much "ad-hoc" and suited to this particular use case, but perhaps with a little modification it could serve your purpose as well.

I uses a 4 million baud serial link with hardware flow control (RTS/CTS lines connected). On top of that it sends packets as defined in RFC 1055 (simple byte stuffing). Each packet's payload is followed by its CRC32.

Possible packet types are defined in dual.h.

The B side can send the following packet types:

DEVICE_CONNECTED (comes with HID report descriptor)
DEVICE_DISCONNECTED
REPORT_RECEIVED
REQUEST_B_INIT

The A side can send the following:

B_INIT (contains configuration, currently only the polling rate override parameter)
RESTART (tells the B side to restart, needed if the polling rate override is changed)

The B side starts by sending REQUEST_B_INIT in a loop until it receives B_INIT. Then it sends the DEVICE_CONNECTED/DISCONNECTED and REPORT_RECEIVED packets as appropriate.

This is just a quick description, but if you're familiar with Arduino you should be able to read the relevant source files and figure out the details. Again, it would probably make more sense to modify the B side a bit to serve the more generic use case. Also keep in mind it only supports USB HID (not MIDI for example).

closed 2022-06-29 by jfedor2

schwarz, 2024-03-21:
@jfedor2 Are MLT04 mice (WMO) supposed to work with the Adafruit Feather RP2040 board or only with two Picos?
Trying the Feather board the mouse (and others of the same make) is not lighting up.

jfedor2, 2024-03-21:
> @jfedor2 Are MLT04 mice (WMO) supposed to work with the Adafruit Feather RP2040 board or only with two Picos? Trying the Feather board the mouse (and others of the same make) is not lighting up.

I don't know, I don't have one to test. There was certainly a time when it didn't work with a single RP2040, but compatibility has improved for some devices since then. But if it doesn't work for you then apparently not for these.

shriramhr, 2024-03-21:
> @jfedor2 Are MLT04 mice (WMO) supposed to work with the Adafruit Feather RP2040 board or only with two Picos? Trying the Feather board the mouse (and others of the same make) is not lighting up.

Hey,

So I had this working with a single pico board a while ago but I think the scroll wheel did not work or some such but after that I tried with 2 picos and all the old mice I tried worked including the ime 3.0, mx518 etc.