jfedor2/hid-remapper #6: Caps Lock and Num Lock indicators are not working

urfdvw, 2022-05-27:
The indicators on the keyboard that are connected are disabled.
Tested on 
- Keychron K4 (both indicators)
- Logitech K360 (This keyboard only have Caps indicator)

jfedor2, 2022-06-06:
Yeah, currently no information about the LED state is passed to the connected device. It is something I would like to add in the future.

fxliang, 2023-02-20:
@jfedor2 I saw this, it might be helpful. 
[Control keyboard leds](https://github.com/sekigon-gonnoc/Pico-PIO-USB/pull/64/commits/a270ab223b694c27d63915d0a8dfb01c1a1cd01a)

columna1, 2023-05-10:
I would love to have this feature, in fact I almost need it. some password prompts I use at work don't have a capslock indicator.
Through a small investigation of my own I think what would be required is parsing of HID_OUTPUT in the device descriptor and then a way of sending states to the keyboard. My limited knowledge of tinyusb, usb in general, and c++ have proven this to be too big of a job for me to do properly so far.

dglaude, 2023-05-11:
What is your plan?
Just transparently sending the LED message from host to device.
Or can we hope for more?

Like
* hid-remapper intercept and react on LED message from the host, like if the user on the host press scroll-lock, hid-remapper could use that in a mapping, to switch Layer (or anything else)?
* Reverse way, maybe hid-remapper could communicate and tell the connected keyboard to turn on that or that LED, as a feedback from an expression, or in a macro?

When playing with LED status in CircuitPython, I discovered a 4th LED that maybe only exist on solaris, the LED_COMPOSE. I would love to have that 4th bit of information, but I am not sure what OS support that.

One funny thing about LED is when you connect multiple keyboard to a Windows machine, if you press Scroll-Lock on one, all the Scroll-Lock go ON. So one device can "communicate" with another device via those LED. Scroll-Lock being the less invasive one (except when using Excell).

jfedor2, 2023-05-11:
Full remapping both ways of course!

Use a keyboard LED to visualize which layer is active.

Use LED state coming from another keyboard, not connected through HID Remapper, as an input.

Swap Caps Lock and Num Lock LEDs. Make them blink!

columna1, 2023-05-11:
Awesome! I am very much looking forward to it!

jfedor2, 2023-05-26:
Please check the latest release! It includes support for keyboard LEDs.

columna1, 2023-05-27:
I have tested it out and it works great 👍. It does everything I want. Thanks a bunch!

closed 2023-05-31 by jfedor2