jfedor2/hid-remapper #7: Support for mousewheel tilt left/right

quadbyte, 2022-06-04:
Really loving the project.
Is there a way to remap the mousewheel tilt (left or right) ?   I'm using an Elecom Huge.
Thanks !

jfedor2, 2022-06-04:
Depends on what kind of mapping you want to achieve. Mapping wheel tilt to another axis (like vertical scroll or cursor X/Y) should work via the "H scroll" input. Mapping one direction to some button and the other direction to some other button, which is probably what you meant, is not currently possible due to how the mappings are defined. It's something I would like to make possible in the future. Perhaps axis->button mappings could work together with the scaling parameter, for example:

H scroll -> button A, scaling=1
H scroll -> button B, scaling=-1

This could also work for things like mapping mouse movement to arrow keys (#5). I will give it some thought.

quadbyte, 2022-06-04:
Thanks for you comment.

This is what I want to achieve : 
Tilt Left -> Ctrl + Shift + Tab  (previous tab in the browser)
Tilt Right-> Ctrl + Tab  (next tab in the browser)

Unrelated, but would be cool to have support for multiple output for a single output. I'm just starting out my config (via the browser tool) and it already start to be cluttered.  May switch to command line tool, but still, would be nice.

[image]

jfedor2, 2022-06-06:
I made an update to the firmware that makes it possible to map positive and negative movement on the same axis to different buttons.

Assuming tilt left/right is equivalent to horizontal scroll on your trackball, it should now be possible to achieve your desired result with a mapping like this:

H scroll -> Left Control, scaling=-1
H scroll -> Tab, scaling=-1
H scroll -> Left Control, scaling=1
H scroll -> Left Shift, scaling=1
H scroll -> Tab, scaling=1

I only tested with V scroll as I don't have access to a device with a tilting wheel at the moment, but if my assumptions about what events it sends are correct then it should also work with H scroll. Anyway, let me know.

As for the UI, I agree, grouping together mappings with the same input sounds like a good idea.

closed 2022-06-06 by quadbyte