jfedor2/hid-remapper #10: Add support for different mappings on different devices

b-, 2022-06-05:
I imagine it wouldn't be too hard to VID:PID pairs and load different configs as a result?

whiskthecat, 2023-05-20:
The main page of the github says multiple devices can be connected through a hub. Did you already partially implement this but just not the per device mapping yet?

jfedor2, 2023-05-20:
> The main page of the github says multiple devices can be connected through a hub. Did you already partially implement this but just not the per device mapping yet?

You can connect multiple devices, you can't have per-device mappings. It's only when you connect multiple devices _of the same type_ this starts to matter.

whiskthecat, 2023-05-20:
Sorry, I am still confused. I guess I will figure it out when my JLCPCB order comes in a week or so, right now I am playing with the BT version. But say I have 2 different devices, don't they both send the same "left, right, forward, back, etc"? So already per-device mapping is relevant? Perhaps the hex code buttons will let them be differentiated? Excellent project, thanks.

jfedor2, 2023-05-20:
Yes! When you connect a mouse and a keyboard, you don't need per-device mappings. When you connect two mice and want to use one for scrolling and the other for moving the mouse cursor, then they would be useful. For now you can use two HID Remappers if you don't need any interactions between the two devices, like switching layers with one that would affect the other.

alanbork, 2024-11-04:
So if I were to connect two nearly identical gamepads I'm going to have trouble mapping one gamepad differently from the other?

jfedor2, 2024-11-04:
> So if I were to connect two nearly identical gamepads I'm going to have trouble mapping one gamepad differently from the other?

No trouble, we've had per-port mappings for some time now.