jfedor2/hid-remapper #9: Layer-switch buttons do not map correctly on alternate layers

b-, 2022-06-05:
Hi,

My specific issue is involving my Kensington SlimBlade. If on layer 0 I map the "back" button to activate L1, and the "forward" button to activate L2, then on L1 I map the "forward" button to send a "forward" HID event, and on L2 I map the "back" button to send a "back" HID event, neither works reliably. 

In other words, what I expect (and am aiming for) is that if I hold the top right quadrant button and click the top left quadrant button, to go back, and vice versa -- if I hold the top left and click the top right button, I want to go forwards. 

I think I've learned most of what I know about HID between this and QMK in the past few weeks so I have no idea how to debug it, but my guess would be that there's some sort of weird conflict where it's interpreting the layer shift commands even though they should be superseded by higher layer commands mapped to those buttons.

KlingonPerson, 2022-07-17:
I set the middle button as set the Layer 2.
and I wanna set

With the Layer 2,
Press the Middle button to set the Layer3 sticky.

> While this is currently the documented behavior ("Layer activating mappings work on all layers, regardless of which layer they are defined on."), I can see how this change could be useful.

KlingonPerson, 2023-01-19:
Hi, This restriction is no longer in effect, and we can now set a each layer key on each layer, right? 
So, issue #9 has been also solved?

> While this is currently the documented behavior ("Layer activating mappings work on all layers, regardless of which layer they are defined on."), I can see how this change could be useful.

jfedor2, 2023-01-19:
It's partially solved. The use case from https://github.com/jfedor2/hid-remapper/issues/9#issue-1261137547 can be configured and works correctly, but sticky layer-activating mappings still don't. I overlooked this when doing the last release and I'm currently looking for a good way of solving it once and for all.

jfedor2, 2023-01-19:
The latest release also handles the sticky layer-activating mappings better.

closed 2023-05-26 by jfedor2