jfedor2/hid-remapper #15: Add Japanese Key Keybindings

IBNobody, 2022-07-12:
Please add the following Japanese keys to your config/web-config.

The 4 Japanese Keys are:

- 0x1C: Convert
- 0x1D: No Conversion
- 0x19: Half-width/Full-width
- 0x15: Katakana/Hiragana

Background:

One trick I would use often with my Elecom mice was to bind the Japanese keys to the buttons, and then use AutoHotkey to trigger off these buttons. Since I had a US keyboard, I never had any conflict.

You can find my old post here...

https://www.reddit.com/r/Trackballs/comments/3zrl8h/elecom_driver_autohotkey_trick/

IBNobody, 2022-07-12:
I can work on this improvement, but it may not be for a few weeks due to vacation.

jfedor2, 2022-07-12:
I'm not sure what the codes you provided are, but they don't seem to be HID usages for those keys.

I don't have access to a Japanese keyboard, but from what I was able to find, these should be the usages for them:

Convert: 0x8a
No Conversion: 0x8b
Half-width/Full-width: 0x35 (same as tilde key)
Katakana/Hiragana: 0x88

You can try importing this JSON config to see if this mapping does what you want:

```
{
    "version": 3,
    "unmapped_passthrough": true,
    "partial_scroll_timeout": 1000000,
    "interval_override": 0,
    "mappings": [
        {
            "source_usage": "0x00090001",
            "target_usage": "0x0007008a",
            "layer": 0,
            "sticky": false,
            "scaling": 1000
        },
        {
            "source_usage": "0x00090003",
            "target_usage": "0x0007008b",
            "layer": 0,
            "sticky": false,
            "scaling": 1000
        },
        {
            "source_usage": "0x00090002",
            "target_usage": "0x00070088",
            "layer": 0,
            "sticky": false,
            "scaling": 1000
        }
    ]
}
```

As mentioned, because the "half-width/full-width" key is the same key as tilde in the USB HID protocol, it won't be possible to map to it separately. Maybe you can use the Japanese backslash/underscore key if you need a fourth one (usage for that one should be 0x00070087).

If the above works for you, we can easily add those keys to the web UI.

IBNobody, 2022-07-13:
Sorry, the codes I provided were the keycodes that the Elecom driver sends when the keys are pressed. 

[image]

I am using Switch Hitter as a way to detect keystrokes.

https://www.majorgeeks.com/files/details/switch_hitter.html

```
11:17.0859  (0x1C, BIOS 0x00) DOWN
11:17.0981  (0x1C, BIOS 0x00) UP -> 123ms
11:19.0286  (0x1D, BIOS 0x00) DOWN
11:19.0393  (0x1D, BIOS 0x00) UP -> 109ms
11:30.0466  (0x19, BIOS 0x00) DOWN
11:30.0601  (0x19, BIOS 0x00) UP -> 136ms
11:32.0269  (0x15, BIOS 0x00) DOWN
11:32.0454  (0x15, BIOS 0x00) UP -> 187ms
11:52.0256 LCtrl (0x11, BIOS 0x1D) DOWN
```

    0x1C: Convert
    0x1D: No Conversion
    0x19: Half-width/Full-width
    0x15: Katakana/Hiragana

I'm at a loss as to why the keystrokes you recommend are different.

jfedor2, 2022-08-30:
Japanese keys should now be supported in the latest version. Please test if you're still interested.

closed 2022-09-06 by jfedor2