jfedor2/hid-remapper #16: Add Layer Cycling

IBNobody, 2022-07-13:
The current setup binds layer shifts to a button. That button can be sticky. This allows for a sniper-assist mode, which is good. However, it only allows for one set of scaling (tied to the layer you are on).

It would be nice if layers could be cycled rather than toggled. 

Press Middle Mouse Button -> Activate Layer 1 with 0.5 X/Y scaling
Press Middle Mouse Button Again -> Deactivate Layer 1, Activate Layer 2 with 0.25 X/Y scaling

jfedor2, 2022-07-13:
If we solved #9, this could be done by configuring a different mapping for the "cycle" button on each layer (switching to the next layer), but we can probably also implement this separately.

jfedor2, 2023-01-19:
With the latest release, this is now possible. You can define the mappings like so:

[image]

(I also added an example that you can click on the config website that sets these.)

closed 2023-05-26 by jfedor2