jfedor2/hid-remapper #22: Question regarding high speed instead of full speed usb for >1khz polling

shriramhr, 2022-08-15:
Hi JF, 

Is there a way to program the pico so that it selects usb high speed instead of full speed? If this is possible then maybe devices can be forced to poll at greater than 1 KHz.

jfedor2, 2022-08-15:
The hardware doesn't support High Speed. The Pico-PIO-USB library doesn't either.

shriramhr, 2022-08-15:
> The hardware doesn't support High Speed. The Pico-PIO-USB library doesn't either.

So technically if this code can be replicated for a different pi board which supports HS usb, it should work right?

jfedor2, 2022-08-15:
In principle, yes. Some parts of the code are specific to the RP2040, in particular the overclocking stuff, but something similar could be done on another chip.

closed 2022-09-06 by jfedor2