jfedor2/hid-remapper #23: why did hid-remapper not report some key when use japanese keyboard?

hiradimir, 2022-08-30:
I trying hid-remapper with Japanese keyboard **https://archisite.co.jp/products/archiss/progres-touch/retro-tkl-jp/**.
But, some below key does not working.

* [¥|] key at left-side on [BS].
* [henkan], [muhenkan] key at side on [SPACE].
* [kana] key at left-side on [Right ALT].
* [\_] key at left-side on [Right SHIFT].

I marked circle not working key in this image. 
[image]

[Unmapped inputs passthrough] configration is true.
[image]

OS-keyboard setting is JIS type ( when press shift-2 appeared ["] )
When I pressed above keys, green LED is flashing on rpi-pico, but does not appear key_code. I am checking key-event by Karabiner-EventViewer.

Is there anything I can do?

jfedor2, 2022-08-30:
I think this is because HID Remapper's output report descriptor doesn't include the codes (usages) for these keys. It should be possible to fix this.

Can you try this: with the keyboard connected, go to the web config tool, open device, then click on the one of the "input" or "output" buttons and scroll to the bottom. After the "Layer 1/2/3" buttons there should be additional buttons with labels like "0x0007008a" or similar. Can you tell me what they are?

jfedor2, 2022-08-30:
Based on this page: https://www.scs.stanford.edu/10wi-cs140/pintos/specs/kbd/scancodes-7.html, I believe the usage codes are as follows:

| key | code |
| --- | --- |
| [¥\|] | 0x89 |
| [henkan] | 0x8a |
| [muhenkan] | 0x8b |
| [kana] | 0x88 |
| [_] | 0x87 |

Currently HID Remapper only supports codes up to 0x73 (on the output side), but it should be an easy fix.

jfedor2, 2022-08-30:
Based on the above assumptions I added support for Japanese keyboard keys in the latest commit (75160dc). They should work in passthrough mode and also it should be possible to use them as mapping inputs and outputs. Please test if you can.

hiradimir, 2022-08-31:
@jfedor2 thank you for your reply

this comment is replying to https://github.com/jfedor2/hid-remapper/issues/23#issuecomment-1231967080

I pickup additional buttons information. 

### captures

<details>

|0x00070000-0x000c0040|0x000c0041-0x000c0124|0x000c0125-0x000c020c|0x000c020d-0x000c023c|
|-|-|-|-|
|[image]|[image]|[image]|[image]|

</details>

###  html from developer tools
https://gist.github.com/hiradimir/240bc105629408d7a9d08767e42a31d7

hiradimir, 2022-08-31:
@jfedor2 
this comment is replying to https://github.com/jfedor2/hid-remapper/issues/23#issuecomment-1232166950

I appreciate the fix.
I tried the commit. I think this version works very well for me.
And also mapping setting test is works well.
[image]

----

By the way, web config tool is not still updated?
I can't see these update at the tool. 
https://github.com/jfedor2/hid-remapper/commit/75160dc43df4e52f5fbc792d473b20af43ad64db#diff-0265bc23f52b243700ba1d9b856bf9145875aeec24235644d6d812f5aaae5a9cR135-R141

[image]

jfedor2, 2022-08-31:
> By the way, web config tool is not still updated?
> I can't see these update at the tool.

Try shift-reloading the website to get the new version.

(I need to sort out the caching so that it's not necessary in the future.)

hiradimir, 2022-08-31:
@jfedor2 sorry, I confirmed these updated.
thanks for your kindness.

[image]

jfedor2, 2022-09-02:
Thank you for verifying, I'll close the issue now.

closed 2022-09-02 by jfedor2