jfedor2/hid-remapper #25: Is it possible to attach a usb male to the hid-remapper?

hiradimir, 2022-09-07:
Is it possible to attach a usb male to the hid-remapper?

I cut the usb extension cable to make the hid-remapper, so the usb male cable is still there.
Is it possible to make effective use of this usb male cable by soldering it to an open gpio pin?

The other day I raised an issue where hid-remapper would not report some keys on the Japanese keyboard.
At that time, I thought that if I could read the console log on the micro usb port of the raspberry pico, I might be able to provide detailed debugging information.

Would it be difficult to handle two usb-io's with raspberry-pico?

jfedor2, 2022-09-07:
What you're proposing is not impossible, but there are (IMO) better options for debugging/diagnostics:

1. We can re-use the Pico's native USB port, the same one that we're using to send mouse/keyboard inputs to the PC. We can add a virtual serial port (called CDC in this context) and it can co-exist with the HID stuff on the same USB port.
2. We can use one the Pico's UARTs along with a separate external USB-UART adapter (often using an FTDI chip).

Personally I'm using option 2 during development, because it's least intrusive and I already have the necessary hardware. But in a situation where I would need to ask a user for debug info to do some troubleshooting I would probably go for option 1, because it would be less of a hassle for them.

hiradimir, 2022-09-07:
Thank you for your reply.

I did not know that I could add a virtual serial port.
Does this option require a code change or just a change in my environment?
If a code change is required, I would not strongly prefer this option.

----

BTW, is it possible to reuse waste of the cut extension usb cable　for make a hid-remapper instead of the micro usb port on the pico for the hid connection to the PC?
I would be happy if you could make a cable-embedded hid-remapper.

jfedor2, 2022-09-08:
> I did not know that I could add a virtual serial port. Does this option require a code change or just a change in my environment? If a code change is required, I would not strongly prefer this option.

Yeah, it requires a code change, it's not currently in the public code.

> BTW, is it possible to reuse waste of the cut extension usb cable for make a hid-remapper instead of the micro usb port on the pico for the hid connection to the PC? I would be happy if I can make a cable-embedded hid-remapper.

Yeah, you can solder the cable to the Pico and use that instead of the micro USB port. There are pads on the bottom of the board labeled TP1, TP2, TP3. TP1 is GND, TP2 is D- and TP3 is D+. You'd also have to connect VBUS (5V).

hiradimir, 2022-09-08:
Thanks for your information.

I tried soldering. It worked perfectly.

My idea was that to use UART1, UART2 etc. 
Your idea was not in my brain.

This uploaded image is reference that using TP1-3 for hid-remapper.
Sorry, my soldering skill poor, but it is work fine.

[image]

closed 2022-09-09 by hiradimir