jfedor2/hid-remapper #26: Possible crystal oscillator issue on custom boards

jfedor2, 2022-12-02:
I suspect some custom boards might have an issue where the crystal oscillator needs a bigger delay at startup.

Attached test build of the firmware has the usual workaround for that issue (PICO_XOSC_STARTUP_DELAY_MULTIPLIER 64).

[remapper_xosc_test.zip](https://github.com/jfedor2/hid-remapper/files/10141488/remapper_xosc_test.zip)

andrewjfreyer, 2023-01-14:
Is this startup delay incorporated into the remapper_board.uf2 in the current release?

jfedor2, 2023-01-14:
No. I never got a confirmation that this is actually an issue, it was just a guess from my other adventures with RP2040 boards.

jfedor2, 2023-02-18:
The delay multiplier is now enabled for all builds in the latest release.

closed 2023-02-18 by jfedor2