jfedor2/hid-remapper #27: pairing with zmk keyboard

claviger-pc, 2022-12-13:
I am able to pair with several devices without problems (Logitech MX keys, MX anywhere 3). I am using a XIAO-ble board. I can also pair with a BT60v1 board running zmk. This board connects, and the lights show that it is receiving input, but there is nothing that gets sent to the host (Windows).

jfedor2, 2022-12-14:
Sure, but currently I'm just using UART serial for that. It's enabled by default (at some verbosity level at least) if you have an adapter lying around.

Not sure what's going on with ZMK, if it's receiving inputs then it would seem the hardest part is working OK. Maybe something with the report descriptor.

claviger-pc, 2022-12-14:
ok, great, I was able to get the physical uart up and that helped to clarify things. In fact, it does not seem that output is coming through the MX keys or the zmk board. Rather, they both connect fine but then there is no output. I can connect a mouse, however (MX anywhere 3) and that works on the host (tried on Windows and ChromeOS).

Any ideas for further testing? Also, I was able to clone and build the regular firmware but there are some steps I am not sure of for setting up the build environment for zephyr.

jfedor2, 2022-12-14:
Yeah, setting up the build environment is kind of a hassle, you can either follow these steps:

https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/gs_installing.html

or use Nordic's Docker image, something like:

```
docker run --rm -v /path/to/hid-remapper:/workdir/project -w /workdir/project/firmware-bluetooth nordicplayground/nrfconnect-sdk:v2.2-branch west build -p always -b seeed_xiao_nrf52840
```

I also just added GitHub workflows for automatic builds so you can just fork and use that.

Setting CONFIG_LOG_MAX_LEVEL=4 in prj.conf should enable dumping report descriptors, if you paste it here I can try and see if there's anything there that we may be parsing incorrectly.

claviger-pc, 2022-12-14:
great! I was able to fork, enable more logging, and hopefully get the necessary info:

*** Booting Zephyr OS build c0689b16ff12 ***  
[00:00:00.002,197] <inf> remapper: main: HID Remapper Bluetooth  
[00:00:00.003,387] <inf> usb_hid: hid_do_status_cb: Device connected  
[00:00:00.003,387] <inf> usb_hid: hid_do_status_cb: Device connected  
[00:00:00.006,469] <inf> usb_hid: hid_do_status_cb: Device suspended  
[00:00:00.006,500] <inf> usb_hid: hid_do_status_cb: Device suspended  
[00:00:00.008,850] <inf> fs_nvs: nvs_mount: 8 Sectors of 4096 bytes  
[00:00:00.008,880] <inf> fs_nvs: nvs_mount: alloc wra: 1, f10  
[00:00:00.008,880] <inf> fs_nvs: nvs_mount: data wra: 1, 240  
[00:00:00.009,002] <inf> sdc_hci_driver: hci_driver_open: SoftDevice Controller build revision:  
                                         6d 90 41 2a 38 e8 ad 17  29 a5 03 38 39 27 d7 85 |m.A*8... )..89'..  
                                         1f 85 d8 e1                                      |....  
[00:00:00.011,474] <inf> bt_hci_core: hci_vs_init: HW Platform: Nordic Semiconductor (0x0002)  
[00:00:00.011,505] <inf> bt_hci_core: hci_vs_init: HW Variant: nRF52x (0x0002)  
[00:00:00.011,535] <inf> bt_hci_core: hci_vs_init: Firmware: Standard Bluetooth controller (0x00) Version 109.16784 Build 2917677098  
[00:00:00.011,901] <inf> bt_hci_core: bt_init: No ID address. App must call settings_load()  
[00:00:00.013,275] <inf> remapper: remapper_settings_set: len=2048  
[00:00:00.017,120] <inf> bt_hci_core: bt_dev_show_info: Identity: D8:94:16:xx:xx:xx (random)  
[00:00:00.017,150] <inf> bt_hci_core: bt_dev_show_info: HCI: version 5.3 (0x0c) revision 0x11fa, manufacturer 0x0059  
[00:00:00.017,181] <inf> bt_hci_core: bt_dev_show_info: LMP: version 5.3 (0x0c) subver 0x11fa  
[00:00:00.108,001] <inf> usb_hid: hid_do_status_cb: Device resumed  
[00:00:00.108,032] <inf> usb_hid: hid_do_status_cb: from suspend  
[00:00:00.108,032] <inf> usb_hid: hid_do_status_cb: Device resumed  
[00:00:00.108,062] <inf> usb_hid: hid_do_status_cb: from suspend  
[00:00:00.108,093] <inf> usb_hid: hid_do_status_cb: Device reset detected  
[00:00:00.108,093] <inf> usb_hid: hid_do_status_cb: Device reset detected  
[00:00:00.152,862] <inf> usb_hid: hid_do_status_cb: Device reset detected  
[00:00:00.152,893] <inf> usb_hid: hid_do_status_cb: Device reset detected  
[00:00:00.224,639] <inf> usb_hid: hid_do_status_cb: Device configured  
[00:00:00.224,639] <inf> usb_hid: hid_do_status_cb: Device configured  
[00:00:00.252,197] <inf> remapper: get_report_cb: report_id=99, 3, len=0  
[00:00:00.252,716] <inf> remapper: set_report_cb: report_id=99, report_type=3, len=2  
[00:00:00.252,899] <inf> remapper: get_report_cb: report_id=99, 3, len=0  
[00:00:00.253,295] <inf> remapper: set_report_cb: report_id=99, report_type=3, len=2  
[00:00:01.180,908] <inf> remapper: scan_filter_match: scan_filter_match address: D5:3E:16:03:BC:CF (random) connectable: yes  
[00:00:01.181,610] <inf> remapper: scan_connecting:  
[00:00:01.286,499] <inf> remapper: connected: D5:3E:16:xx:xx:xx (random)  
[00:00:01.682,220] <inf> remapper: security_changed: D5:3E:16:xx:xx:xx (random), level=2.  
[00:00:02.154,663] <inf> remapper: discovery_completed_cb:  
[00:00:02.199,707] <inf> remapper: hogp_ready_work_fn: hogp_ready

jfedor2, 2022-12-14:
I don't think you changed the log level successfully.

Just change CONFIG_LOG_MAX_LEVEL=3 to 4 in the second line of prj.conf.

Then there should be something like this in the log:

```
[00:00:03.055,511] <dbg> remapper: main: incoming_descriptor
```

With a hex dump following.

claviger-pc, 2022-12-14:
ok, sorry about that, I didn't realize it was already defined above in the prj.conf and it cannot be changed below. Here it is with the higher level of logging. Actually, it seems to crash the xiao. When I bond with it and send info, which I take would be a sign of some problem with parsing. The MX Keys, on the other, hand does really work, I just realized that the zmk board was causing the xiao to crash and so nothing was working.

[00:00:00*** Booting Zephyr OS build c0689b16ff12 ***  
[00:00:00.002,197] <inf> remapper: main: HID Remapper Bluetooth  
[00:00:00.003,387] <inf> usb_hid: hid_do_status_cb: Device connected  
[00:00:00.003,417] <inf> usb_hid: hid_do_status_cb: Device connected  
[00:00:00.006,500] <inf> usb_hid: hid_do_status_cb: Device suspended  
[00:00:00.006,500] <inf> usb_hid: hid_do_status_cb: Device suspended  
[00:00:00.007,690] <inf> fs_nvs: nvs_mount: 8 Sectors of 4096 bytes  
[00:00:00.007,720] <inf> fs_nvs: nvs_mount: alloc wra: 1, dd0  
[00:00:00.007,720] <inf> fs_nvs: nvs_mount: data wra: 1, 564  
[00:00:00.007,843] <inf> sdc_hci_driver: hci_driver_open: SoftDevice Controller build revision:  
                                         6d 90 41 2a 38 e8 ad 17  29 a5 03 38 39 27 d7 85 |m.A*8... )..89'..  
                                         1f 85 d8 e1                                      |....  
[00:00:00.010,284] <inf> bt_hci_core: hci_vs_init: HW Platform: Nordic Semiconductor (0x0002)  
[00:00:00.010,314] <inf> bt_hci_core: hci_vs_init: HW Variant: nRF52x (0x0002)  
[00:00:00.010,345] <inf> bt_hci_core: hci_vs_init: Firmware: Standard Bluetooth controller (0x00) Version 109.16784 Build 2917677098  
[00:00:00.010,711] <inf> bt_hci_core: bt_init: No ID address. App must call settings_load()  
[00:00:00.012,359] <inf> remapper: remapper_settings_set: len=2048  
[00:00:00.016,815] <inf> bt_hci_core: bt_dev_show_info: Identity: D8:94:16:D7:C1:BC (random)  
[00:00:00.016,845] <inf> bt_hci_core: bt_dev_show_info: HCI: version 5.3 (0x0c) revision 0x11fa, manufacturer 0x0059  
[00:00:00.016,876] <inf> bt_hci_core: bt_dev_show_info: LMP: version 5.3 (0x0c) subver 0x11fa  
[00:00:00.104,888] <inf> usb_hid: hid_do_status_cb: Device reset detected  
[00:00:00.104,888] <inf> usb_hid: hid_do_status_cb: Device reset detected  
[00:00:00.104,919] <inf> usb_hid: hid_do_status_cb: Device resumed  
[00:00:00.104,919] <inf> usb_hid: hid_do_status_cb: Device resumed  
[00:00:00.149,505] <inf> usb_hid: hid_do_status_cb: Device reset detected  
[00:00:00.149,505] <inf> usb_hid: hid_do_status_cb: Device reset detected  
[00:00:00.253,295] <inf> usb_hid: hid_do_status_cb: Device configured  
[00:00:00.253,326] <inf> usb_hid: hid_do_status_cb: Device configured  
[00:00:00.280,517] <inf> remapper: get_report_cb: report_id=99, 3, len=0  
[00:00:00.281,188] <inf> remapper: set_report_cb: report_id=99, report_type=3, len=2  
[00:00:00.281,219] <dbg> remapper: set_report_cb:  
                                   63 01                                            |c.  
[00:00:00.281,463] <inf> remapper: get_report_cb: report_id=99, 3, len=0  
[00:00:00.281,860] <inf> remapper: set_report_cb: report_id=99, report_type=3, len=2  
[00:00:00.281,860] <dbg> remapper: set_report_cb:  
                                   63 05                                            |c.  
[00:00:01.030,334] <dbg> remapper: process_bond: D5:3E:16:xx:xx:xx (random)  
[00:00:01.030,548] <dbg> remapper: scan_setup_filters: scanning for bonded peers only  
[00:00:01.031,219] <dbg> remapper: scan_start: Scanning started.  
[00:00:01.235,473] <inf> remapper: scan_filter_match: scan_filter_match address: D5:3E:16:xx:xx:xx (random) connectable: yes  
[00:00:01.236,175] <inf> remapper: scan_connecting:  
[00:00:01.341,888] <inf> remapper: connected: D5:3E:16:xx:xx:xx (random)  
[00:00:01.737,030] <inf> remapper: security_changed: D5:3E:16:xx:xx:xx (random), level=2.  
exit

claviger-pc, 2022-12-14:
this is from mx mouse, I do not get one like this for the zmk keyboard:
[image]

jfedor2, 2022-12-14:
You got me curious so I did some digging. It seems that we're having a problem with this part of ZMK's report descriptor:

```
0x05, 0x0C,        // Usage Page (Consumer)
0x09, 0x01,        // Usage (Consumer Control)
0xA1, 0x01,        // Collection (Application)
0x85, 0x02,        //   Report ID (2)
0x05, 0x0C,        //   Usage Page (Consumer)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x0F,  //   Logical Maximum (4095)
0x19, 0x00,        //   Usage Minimum (Unassigned)
0x2A, 0xFF, 0x0F,  //   Usage Maximum (0x0FFF)
0x75, 0x10,        //   Report Size (16)
0x95, 0x06,        //   Report Count (6)
0x81, 0x00,        //   Input (Data,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              // End Collection
```

Likely running out of memory because we're making a separate entry for every one of those 4096 usages. I will have to think about how to improve this.

claviger-pc, 2022-12-14:
okay, let me know if there is something you want me to test. You could also run zmk on a spare xiao ble if you want to test yourself. The people on the zmk discord are quite friendly too. Perhaps it would be worth considering integrating your efforts into zmk, then you would have all the power of zmk's mapping and features? Just an idea, but I am sure the community would be delighted to have a great developer like you around.

jfedor2, 2022-12-16:
I made an update that I think could fix this (502837f7c5f1fb96ced2ceb711b3fa1b57257694). Please try the latest release if you have a moment.

claviger-pc, 2022-12-16:
I will try as soon as possible, but I just left for the holidays and did not bring a zmk keyboard or Bluetooth board with me. I should be back on January 11 and will test then. Thank you for your efforts, this is very exciting!!

claviger-pc, 2023-01-11:
I am typing this using a ZMK keyboard paired to a Xiao nrf52840, which means that basic functionality is working and it no longer crashes when connecting to a ZMK board.

jfedor2, 2023-01-11:
Awesome. Thank you for testing.

closed 2023-01-19 by jfedor2