jfedor2/hid-remapper #28: Question: Support for wired Apple Magic Keyboard?

GhostInTheSteiner, 2022-12-18:
I'm currently considering to use your guide to use an Apple Magic Keyboard on Windows. I could mostly solve mapping problems with autohotkey so far, but for RDP sessions or virtual machines these autohotkey scripts won't work for obvious reasons.

My plan would be to buy the v1 board from JLCPCB and then use it on a Windows based OS.

I'm somewhat worried because the Magic Keyboard is custom tailored for Mac computers, although it works on Windows just fine with some swapped buttons like Alt and Win.

Would you say your boards are compatible with the Magic Keyboard? Or is there something I should take into consideration before ordering the boards?

closed 2022-12-19 by GhostInTheSteiner

peterburk, 2024-08-22:
I tried building it today (both the single Pico and dual Pico variants) and the Magic Keyboard A1843 does not work :(

I'll be looking into the reasons why, and trying to find a solution.

peterburk, 2024-08-29:
The problem is in enumeration in Pico-PIO-USB, so I've raised an issue with that project. 

[https://github.com/sekigon-gonnoc/Pico-PIO-USB/issues/135](https://github.com/sekigon-gonnoc/Pico-PIO-USB/issues/135
)

peterburk, 2024-08-29:
Trying with the `host_cdc_msc_hid` example, I had some success!

```
 TinyUSB Host CDC MSC HID Example
HID device address = 1, instance = 0 is mounted
HID Interface Protocol = None
HID has 2 reports 
HID device address = 1, instance = 1 is mounted
HID Interface Protocol = Keyboard
A device with address 1 is mounted
```

Typing on the keyboard then emits serial packets. 

However, the CAPS lock LED doesn't turn on when it's pressed.