jfedor2/hid-remapper #30: Mapping one key + modifier to multiple keys possible

pkockaer, 2022-12-20:
This project is a very elegant solution to remap keyboards at an affordable cost, and I am eager to buy a pi board and test it.

I would like to attach it to a BÉPO keyboard so that it looks like an AZERTY one.

The problem I foresee is that upper case accented letters such as É would require to input the dead accent ´ followed by the uppercase E: É = ´ + E.
Would it be possible to implement this easily (one key + modifier maps to 2 keys) ? Do you think it would require a lot of changes and affect many files? 

Other combinations will also be needed, for example to map Ç. On windows, this requires to input ALT+0199, mapping therefore one key + shift modifier to 4 keys + alt modifier.

I could devote some time to make this work, but I would appreciate if you can tell me how hard this looks to you and where to start.

jfedor2, 2022-12-22:
Hi, thank you for taking interest in the project.

The functionality you're describing is certainly something I would like to add, but I haven't got around to it yet.

Currently, with the exception of the sticky mappings, the current output state (the inputs sent to the computer) is directly derived from the inputs read from the peripherals (mice, keyboards). There's no way to say send this and then in the next report send something else.

It seems that we need a way to define macros and a way to execute them.

Each macro could be a list of usage-value pairs, allowing some flexibility, for example the macro for for É would be:

```
´ : 1
´ : 0
E : 1
E : 0
```

The macro for Ç would be:

```
alt: 1
0 : 1
0 : 0
1 : 1
1 : 0
9 : 1
9 : 0
9 : 1
9 : 0
alt : 0
```

They could also conceivably include things like mouse movement. We could have a macro for double click. Do we need something like a pause in the middle of the macro definition?

Internally we'd probably have some list of macros and then mappings could have "trigger macro X" as the target usage, similarly to how layer switches are done currently.

Then of course when the user presses the button that triggers the macro we need to be able to send those inputs in consecutive reports. We could either generate all the inputs at that time and put the generated reports in some queue (there's currently a queue for outgoing reports, but it's fixed size and short, because it's for situations when the user is moving the mouse and pressing keys at the same time and we sometimes generate reports more quickly than we can send them). Or we could have some flag that says we're currently executing macro X and we've done N out or M steps in it and generate the reports in real time, every time there's a tick_pending.

We have to keep in mind that the user might for example be moving the mouse at the same time we're playing back a macro. This won't be a huge issue as the macro will usually be a few letters, probably always less than, say, 100 reports generated in total, but still we need to consider if we want other inputs that come in during this time to be interleaved with the reports generated from the macro (probably yes, to avoid choppy mouse movements).

Then of course there's the issue of how to present it all to the user and let them configure it. Do we have interactive macro recording (first version can definitely do without it)?

There's probably some other things that I'm missing right now, like what to do about auto-repeat. If the user has a key that sends "É" then they probably expect to type "ÉÉÉÉÉÉ..." if they keep pressing it. But if they have a mapping that makes the right button generate a double click then they probably want it to generate exactly one double click every time, even if the button is held a little longer.

pkockaer, 2022-12-23:
Thank you very much for your detailed in put.

This motivates me to buy the hardware and start to test the remapper.

Once this will be done, I will try to define simple macros and come back to you if I can progress on that. 

As I was focused on the keyboard aspect, I did not pay attention to the possible interactions with the mouse input.
Concerning the dead keys, and probably also the ALT+xxxx combinations, this should not be an issue, as I can move my mouse between the two key strokes without any problem (at least on linux).

jfedor2, 2023-01-15:
Check out the latest release, it now has macros.

pkockaer, 2023-01-16:
Thank you very much for implementing that feature. I have the device ready and will build and test the new version.
I already enjoy the 2D interface to edit macros.

pkockaer, 2023-01-17:
I have made some tests and it works pretty well.
This is really great, as it will allow to build a hardware-coded BÉPO keyboard. 
Thank you again for implementing this project. This will save me a lot of efforts. 
I'll post the config file when it is ready.

closed 2023-01-19 by jfedor2