jfedor2/hid-remapper #32: Config tool: allow keypress to select Input

aszymanik, 2022-12-29:
I was hoping for a possible enhancement with the web config tool- I have some non standard keys (euro, yen, $) on my keypad and the selection list for input choices has hundreds of hex options. Without going through all of them through trial and error, my thought is to allow the input box to accept a keypress from the keypad then move to the next Output selector for remapping. I'm able to capture the scan codes with the linux tools showkey and xev but their output does not correlate with the options I see available in the input box. 

This is the keypad I'm using: https://www.amazon.com/dp/B09TYQ2TKZ

Thank you!

aszymanik, 2022-12-29:
Examples of unknown keys as reported from "showkey -s"
Button A: 0x45 0xc5 0x45 0xc5
Button B: 0x45 0xc5 0x38 0x4b 0x52 0xb8 0xcb 0xd2 0x45 0xc5
Button C: 0x45 0xc5 0x1d 0x38 0x2d 0x4b 0x9d 0xb8 0xad 0xcb 0x45 0xc5

jfedor2, 2022-12-29:
Interactive remapping is definitely on my list (in fact I think it's listed in the "goals" section).

To get an idea what HID usages the keyboard is actually sending I'd recommend using usbhid-dump or hid-recorder from the [hid-tools](https://gitlab.freedesktop.org/libevdev/hid-tools) project.

With usbhid-dump you will probably want to put the dumped HID descriptor through something like [this parser](https://eleccelerator.com/usbdescreqparser/) to make sense of it. hid-recorder has a built-in descriptor parser.

From what you posted it's possible that the keypad is actually sending multiple keystrokes when those keys are pressed (which wouldn't be optimal for use with HID Remapper). If you post the output from one of the above tools (HID report descriptor and then HID reports sent for each key), we can try and figure it out.

aszymanik, 2022-12-29:
Sorry for this wall of text,  I think you're right though- it's sending out multiple keys at the same time. It's just a numpad so I was thinking if I could capture the first keycode I could ignore the rest (and uncheck "unmapped inputs passthrough")

[out.txt](https://github.com/jfedor2/hid-remapper/files/10322495/out.txt)

jfedor2, 2022-12-29:
It looks like you had the numpad connected through the remapper when you dumped these inputs. Try connecting it directly to the computer, it might be sending some codes (vendor-specific maybe) that are not passed through.

Your plan might work for the yen key as it seems to be sending ctrl-alt-X, among other things, so you could just map "X". With the others it might be tricky, for example the double zero key is really sending zero twice. The numlock/dollar key is just sending numlock (twice for some reason).

If the numpad is depending on the numlock LED state for its logic, that might be throwing it off because the remapper doesn't pass the LED state from the computer (another thing on my list).

aszymanik, 2022-12-30:
Interesting about the LED state- I noticed the $ key would sometimes send a value and other times nothing at all as if the button wasn't pressed. I'll dig more into that.

Unfortunately, I already took apart the dongle and soldered it directly to the pico. I won't be able to directly connect it to the computer now :(

I was able to map:
Euro and Yen: LeftAlt
Doesnt seem like the LeftControl or X is captured.  

Not working:
$: can map to Num Lock but seems like the key keeps repeating and sends characters until I turn off the num luck with the actual button
00: Is sending 0 twice. So remapped 0 and 00 are doing the same thing (but with a repeat). Oh well.

jfedor2, 2023-05-26:
It's not interactive remapping, but you can now see what inputs your device is sending in the "Monitor" tab.