jfedor2/hid-remapper #34: Output non valid keys - custom output keyId

Dschogo, 2023-01-03:
Hey there,

Im thinking of using this in combination of AHK (autohotkey) to automate some stuff. 
Why? I got an wireless numpad and want to use it as a macro pad - not just for key-combos/shortcuts also for interacting with webrequests/home assistant/dlls etc. 

From my understanding the remapper can technically send anything to the pc (in terms of "keys"), so my idea would be to remap lets say Num1 to an unused keyid. AHK can read raw hid/input data and handles then the rest.  (no other program will do anything since it doesn't understand the key or doesn't even listen to them)

That said, would it be possible to configure "custom" keyid's in general (output)?
Does it already work with editing the json? (i've seen code changes for adding track balls, but that was for input not output)
Furthermore, if all that works, for ease of use add that "use custom keyid" in the webtool for others / ppl that arent comfortable directly editing jsons.

PS: awesome project :D

jfedor2, 2023-01-03:
Right now the set of outputs that can be sent is fixed. Mapping to an unknown ID via JSON will just result in nothing being sent. But there are some keys like F13-F24 that are unlikely to be used for anything on your computer so you could use those for custom mappings in AHK.

Dschogo, 2023-01-03:
Thanks for the very quick reply. 
Alright, that would've been my b-plan. I'll probably also use the japanese buttons, since the numpad has 18 buttons to remap. 
Apart from Adobe products i dont think any other has mapping for F11-F24.

Then it can be seen as a future feature request to enable advanced/large macrokeyboards (thinking of a fullsize keyboard - handy for video editors etc)