jfedor2/hid-remapper #37: Board v1 - Side A flashing Side B required each power cycle

andrewjfreyer, 2023-01-14:
This is a bit unusual, but for the custom board v1 flashing Side B via Side A seems to work just fine. Copy remapper_board.uf2 over to Side A in boot select mode, load config page, press "load side B." Thereafter, the config page updates its UI to say "Flash succeeded, please unplug and plug in." Unplug and plug back in, device works great. 

Until the *next* power cycle. Board v1 is unplugged sometime later, and a re-flash of Side B is required before remapping works again. 

Repeatable on my end across four different v1 boards.

jfedor2, 2023-01-14:
This is weird. I don't have an explanation.

andrewjfreyer, 2023-01-14:
Is there some logging or debugging that I could provide?

andrewjfreyer, 2023-01-14:
Well, whatever was causing the issue the release you made about an hour ago fixes it!

closed 2023-01-14 by andrewjfreyer