jfedor2/hid-remapper #40: Bug: Low-speed USB devices are not recognized

kelnishi, 2023-01-31:
Keyboards operating in USB Low-Speed 1.5Mbps mode are not recognized.

jfedor2, 2023-01-31:
If you have a moment, please try this build. It uses an updated version of the pico_pio_usb library and in my tests it helped with some devices. (Unfortunately it also seems to break some other devices, which is why I haven't put this in the normal release yet.)

[remapper-newer-pio-usb.zip](https://github.com/jfedor2/hid-remapper/files/10551049/remapper-newer-pio-usb.zip)

kelnishi, 2023-02-01:
This newer firmware does indeed work with low-speed (1.5Mbps), however it no longer works with high-speed (12Mbps). The latest host_hid_to_device_cdc.c from Pico-PIO-USB does recognize both speed devices and can successfully hot plug both in succession.

jfedor2, 2023-02-01:
Interesting. I need to look into this. Thank you for testing.

jfedor2, 2023-02-03:
So the difference between HID Remapper and the host_hid_to_device_cdc example is that HID Remapper uses the Pico-PIO-USB library directly, while host_hid_to_device_cdc uses it as a backend to tinyusb. I've tried this setup before, but I've ran into some issues, I don't remember the details. I've tried it again with the current versions of tinyusb and Pico-PIO-USB and everything seems to work fine and it does work with some devices that the current single Pico version doesn't. In fact it seems to work with all the devices I have, except some hub that has issues even with the dual Pico version.

Here's a single Pico test build that uses Pico-PIO-USB as a backend to tinyusb if you're up for some more tests.

[remapper_pio_tinyusb_backend.zip](https://github.com/jfedor2/hid-remapper/files/10581923/remapper_pio_tinyusb_backend.zip)

kelnishi, 2023-02-03:
This one successfully connects to all the devices I have (HS keyboard, LS keyboard, HS wireless mouse dongle).
I do see an error when using the configuration tool however:
[image]
This error happens when saving the configuration to device. Once it's in this state, further configuration is not possible and a hard reset (via the pico reset button) is required. The configuration does appear to make it to the board before the error though, so once the pico is rebooted, the remappings are present.

jfedor2, 2023-02-03:
Hmm this kind of sounds like the issues I had with this setup before, but weirdly I'm not experiencing them now.

petrm, 2023-02-06:
> So the difference between HID Remapper and the host_hid_to_device_cdc example is that HID Remapper uses the Pico-PIO-USB library directly, while host_hid_to_device_cdc uses it as a backend to [...]

This build works for me with Kinesis Freestyle 2. The default release doesn't send any keys.

jfedor2, 2023-02-06:
@petrm Do you see any issues when clicking "save to device" on the web config interface with this build?

jfedor2, 2023-02-06:
@kelnishi Do you get the error every time you save the settings to the device? I'm tempted to put this change in the regular release, but not with this problem, obviously.

kelnishi, 2023-02-07:
The NotAllowedError issue when saving the mapping configuration happens 100% for me. In chrome for Windows and macOS. I'm running on an rp2040-zero board without debug pins exposed. I can solder up some other boards with TTL pins to see if I can get more debug for you if that will help.

kelnishi, 2023-02-07:
I wonder if TinyUSB is causing Chrome to be more restrictive with sendReport. Seems like access to HID devices like keyboards is intentionally handicapped https://github.com/WICG/webhid/issues/12#issuecomment-645487885

jfedor2, 2023-02-07:
That's not it, on the computer-facing side all versions use tinyusb, this change only affects the mouse-facing side. And if reading the configuration works, then it means sending reports works. The NotAllowedError can be misleading, it also happens when the device crashes for some reason and stops responding at all. In this case it looks like it crashes when writing the configuration to flash.

It's good that you mentioned that you're using an RP2040-Zero board and not a Pico. I will test with an RP2040-Zero and see if I'm able to reproduce. Usually they work the same, but maybe a different flash chip makes the write take a little longer or something.

kelnishi, 2023-02-07:
I've done some more tests and I've narrowed down the crash/NotAllowedError.
If a HS (12Mbps) USB keyboard or mouse is plugged into the board when mappings are saved, the board will freeze and/or NotAllowedError will present. LS devices do not seem to be affected. 

The problem occurs on all the boards I've assembled:
- Raspberry Pi Pico
- RP2040-Plus
- RP2040-Zero

Both HS and LS devices work correctly on all the boards.
Saving mappings does succeed even when the board freezes, a hard reset will get it functioning again.
Unplugging the HS devices prior to saving mappings will allow the mappings to be saved without freeze/crash. The device can then be re-plugged after saving and the board will continue to function normally.

petrm, 2023-02-08:
> @petrm Do you see any issues when clicking "save to device" on the web config interface with this build?

No issues, I am using RP2040-Zero.

jfedor2, 2023-02-08:
Oh, so it's not board-specific, but depends on what device is connected. What devices are you using specifically, @kelnishi ? I don't think it's as simple as low-speed vs. full-speed, at least I'm not able to reproduce it with my FS devices.

jfedor2, 2023-02-08:
@kelnishi

OK, so this is a bit of a shot in the dark as I'm not able to test it on my side, but if you haven't had enough yet, please try this build and see if it makes any difference.

(The change is that I'm no longer disabling interrupts before writing to flash - it shouldn't be necessary on the single Pico version as the code is always running from RAM anyway.)

[remapper_do_not_disable_interrupts.zip](https://github.com/jfedor2/hid-remapper/files/10686908/remapper_do_not_disable_interrupts.zip)

jfedor2, 2023-02-08:
Oh, cool! Thank you for all the testing.

closed 2023-02-08 by kelnishi