jfedor2/hid-remapper #42: Firmware "lost" after disconnect on rp2040-zero clone

ztxdk, 2023-02-16:
Hi

I have a couple of clones (from ali) of the rp2040-zero.
When I load the uf2 it works fine. But if I pull the connection to the pc and reconnect it's not found by the pc (windows) and I have to reset with the boot button held and get an empty drive (there's just the html/txt file with board info)..
When I load the firmware it works again and have the settings made earlier.
It's the same on both boards.

I have some bigger board (with all the pins and 16mb flash, also from ali) and they work fine after power cycle.

Does anybody know how to make it work?

/martin

jfedor2, 2023-02-16:
This kind of sounds like the crystal oscillator problem that some RP2040 boards have.

Can you try this firmware build and see if it helps?

[remapper_xosc_test2.zip](https://github.com/jfedor2/hid-remapper/files/10759858/remapper_xosc_test2.zip)

ztxdk, 2023-02-16:
That was quick! :-)

But that did it. It works on both of the boards I have! Everything works as on the other bigger boards!

jfedor2, 2023-02-18:
This fix is now in the latest release.

closed 2023-02-18 by jfedor2

ztxdk, 2023-02-18:
Thats awesome! Thank you very much!