jfedor2/hid-remapper #43: Output Key order

Donwick89, 2023-02-18:
Hello,
	I think this project is amazing. It solves a problem that many have trouble with and go to great lengths to fix.  @jfedor2  did a great job.

 I created a dual-Pico running the latest r2023-02-17 as of today. 
 
 
 My question is about the order of output with certain keys. I am having trouble getting key combinations to output in the correct order. I am using F17 & C as an example below. 
 
 In the documentation, using two mappings, the example for LCtrl + C yields as follows:

HID-remapper Configuration:
  Input		Output
	B		Left Control
	B		C
 
 
Output (Recorded by Autohotkey) 
Up/Down  time 	Key 
down	1.73	LControl       	
down	0.00	c              	
up		0.11	LControl       	
up		0.00	c 

The above functions exactly as it was entered. However, if you use F17 (or any fn key), it outputs differently. Whatever the second key is, it is always pressed down first. This seems to happen with all fn keys, and most non-letter keys. 

HID-remapper Configuration:
  Input		Output
	B		F17
	B		C
 
 
Output (Recorded by Autohotkey)
Up/Down  time 	Key
down	 3.77	c              	
down	 0.00	F17            	
up		 0.13	c              	
up		 0.00	F17  

I have tried inverting the mappings in the configuration tool. I have also tried adding the 'nothing' value in different combinations.

I also tried different combinations in the Macros. The macros  output F17Down F17Up CDown Cup

I cannot get the Fn key to output first. 

Can anyone help me get my configuration such that the keys fire in the desired order? Thanks!

jfedor2, 2023-02-18:
When there are two mappings with the same input and two different keys as output, they are sent simultaneously (in the same HID report). The order of the mapping definitions doesn't matter. How the software interprets it is another matter.

If you want to ensure a certain order, you should be able to do it with macros. Do you want the order to be:

F17-down
C-down
C-up
F17-up

?

If so, a macro like this should do the trick:

[image]

Donwick89, 2023-02-20:
Thank you, that worked perfectly! 

Are there any future plans to expand the number of available macros?

jfedor2, 2023-05-26:
The latest release bumps the number of macros to 32.

closed 2023-05-26 by jfedor2