jfedor2/hid-remapper #45: Can't pair on NRF52840. No Bluetooth device visible

Warpshock, 2023-02-26:
Flashed the appropriate firmware on Seeed nRF52840. 
Solid blue light when plugged in. 
Successfully connected through web tool and clicked pair. 
No bluetooth signal visible to laptop or cell phone (S22 Ultra).
I tried nRF Connect app and still don't see it.
Flashed SEEED "HID Keyboard Demo" and that works correctly. [Wrote down MAC address]
Reflashed hid-remapper, but still can't find it.

Can you help?
Thanks!!

jfedor2, 2023-02-26:
There might be some confusion as to what the Bluetooth version (currently) does.

You connect it to your computer via USB and then you connect a mouse or keyboard to it via Bluetooth, not the other way round.

Warpshock, 2023-02-27:
That's how I thought it worked. I failed to mention that I tried pairing with a Logitech Harmony Keyboard and with a Logitech MX 3 Mouse (both) BLE. I tried both the web button and shorting pin 0 to ground for several seconds. Neither caused the solid blue led to change. Clearing all devices did cause a flash. 

I tried thought it should be visible to the nRF Connect app, even if it was not broadcasting its name.

jfedor2, 2023-02-27:
Did you put your peripherals in pairing mode? Can you see *them* in nRF Connect?

closed 2023-05-26 by jfedor2

jfedor2, 2023-07-24:
Can you install the nRF Connect app on your phone and see if the keyboard is visible there?

orbit3r, 2023-07-26:
I did, the keyboard actually doesn't show up on scan despite showing up and being pairable with my Desktop pc using an intel ax200 wifi/bluetooth card (I put it in pairing mode using another one of the 3 pairing slots)
I assume the issue is with my keyboard, any idea on how to troubleshoot this?

jfedor2, 2023-07-26:
It's not that there's anything wrong with your keyboard, but if you can't see it in nRF Connect then it most likely uses Bluetooth Classic, not Bluetooth Low Energy. HID Remapper only works with Bluetooth LE devices (currently).