jfedor2/hid-remapper #48: Hold instantly activates when using macros

blakie1225, 2023-03-11:
I have 2 mappings setup for one input key, one is for a tap and the other for holding the key down.  However when the held is set to a macro, the macro instantly activates instead of waiting for the held delay. I’ve tested the held key as another action which doesn’t instantly activate but when setback to a macro the held delay is ignored. Am I doing something wrong or is this a bug

jfedor2, 2023-03-11:
This can probably be fixed. The macro would still only trigger once (no auto-repeat), I assume that's what you expect.

blakie1225, 2023-03-12:
Yep thats exactly what I want, thanks heaps if you can make this happen. Also kudos for this program, I'm using it on my media pc with a usb remote and it has let me control heaps of tasks just with a basic remote.

jfedor2, 2024-01-10:
I know it's been some time, but in the latest release "hold" macro mappings should now work correctly!