jfedor2/hid-remapper #49: Question: How are you maintaining DPI / mouse speed?

D3VL-Jack, 2023-03-12:
Hi there! 

First off, thankyou for this awesome project, it's been extremely useful to use as a reference and check my logic against. 
Some background, I'm building a tool similar to this but with a completely different application. 

I've built a complete HID parser that will hopefully be open sourced soon and have confirmed data being returned from it and it's translation layer is 100% correct in line with the HID spec, though when sending a report back to the pc the mouse speed is almost 2x over the same physical distance.
This behaviour is not seen with this project on the exact same hardware, and I can't see how you've managed it.

In my fruitless efforts to debug this, I've gone as far as copying your mouse descriptor entirely (with the removal of AC Pan) to confirm that's not the issue, and with that in place using WireShark, I can observe the following report from this project `01000200feff00000000`
```
ID BB XXXX YYYY WWWW AAAA
01 00 0200 feff 0000 0000
```
And from my own code observe the following report `fa000700feff0000`
```
ID BB XXXX YYYY WWWW
fa 00 0700 feff 0000
```
I am triggering a report to be sent to the pc on every report received to the Pico, and as such the Pico is basically a transparent layer with no modification to the data.

The only thing I could see remotely close to speed would be the `Report ID (REPORT_ID_MULTIPLIER)` field you define in the report, but this not only doesn't affect the X, Y Usage Locals, but is only used throughout your code to change the scroll wheel.

Any tips or pointers where to look to solve this would be immensely appreciated, the past week trying to get my head around the HID Descriptor format has been painful! 

Thankyou for your time, I appreciate this is slightly off topic for the repo, you're the only person I can see who has successfully tackled this!

Jack

-----------------

Some additional notes:
- Mouse for testing is an Intellimouse Pro NGX-00012 with it's DPI set in windows to 1400
- Only one Pico in use
- TinyUSB also has a CDC interface on the device end
- Attached are 2 pcap files in a zip, one showing this project, the other my implementation and a copy of my descriptor.
  - [remapper-pcaps.zip](https://github.com/jfedor2/hid-remapper/files/10952348/remapper-pcap.zip)
  - [hid_descriptor.txt](https://github.com/jfedor2/hid-remapper/files/10952345/hid_descriptor.txt)

jfedor2, 2023-03-13:
Show us the code.

D3VL-Jack, 2023-03-13:
Gist is here, 
https://gist.github.com/D3VL-Jack/ffc6de8f65a8ea6a850aed947ae0dffd

I've added what I believe to be relevant, though please let me know if there's anything else! 
Assuming by your response this isn't an issue you ran into yourself? 

Cheers!

jfedor2, 2023-03-13:
One thing that jumps out is that your version sends the reports roughly every 6ms, apparently because that's how often it's receiving them. It's only natural that you'll get 6x bigger numbers from the mouse compared to 1ms polling. That in itself shouldn't result in different mouse speed on the computer unless acceleration is somehow affecting it.

closed 2023-05-26 by jfedor2