jfedor2/hid-remapper #54: Elecom Huge Wired does not function with 2 Pico variant

ww123td, 2023-04-05:
It seems that the Elecom Huge wired version is not working with the device. I had only tested it with the 2 Pico variant. Upon connection, the LED on the mouse would sometimes blink (when connected directly to a device it would blink twice) and no input could be read. The status LED on neither Pico would blink.
I would like to assist in debugging the device, but since I only have limited hardware knowledge I would need instructions on where and what information I can provide to help.
I tried using win-hid-dump to get the HID descriptors but it seems to always crash. But I did see 4 entries created by this mouse so it could still be related to #18

jfedor2, 2023-04-05:
Can you try dumping the report descriptor using Wireshark? From what I remember the trick is to unplug and re-plug the device once the capture is running.

ww123td, 2023-04-05:
Hello, here is the dump file from Wireshark. I tried to filter out the useful entries and saved in a separate file, but in case I had omitted important info I have attached the full dump from my session as well.
[Elecom Huge dump.pcapng.gz](https://github.com/jfedor2/hid-remapper/files/11162851/Elecom.Huge.dump.pcapng.gz)
[Elecom Huge dump full.pcapng.gz](https://github.com/jfedor2/hid-remapper/files/11162852/Elecom.Huge.dump.full.pcapng.gz)

jfedor2, 2023-04-05:
Thanks. It doesn't seem that the descriptor is the issue in this case. Might be something lower level. We could probably increase TinyUSB debug level and see if it outputs anything relevant. That would require a USB-to-TTL adapter to read debug output from the Picos.

ww123td, 2023-04-05:
Hi, thanks for your reply. I would like to help but it would be great if you could point me to a step by step guide on how to set up the debug environment, as I'm very inexperienced with hardware debugging. Would I need to modify and compile the Pico firmware too? Thanks for the help.

jfedor2, 2023-04-05:
I can provide you with some test firmware if you have the hardware to read the serial output from the Picos.

ww123td, 2023-04-05:
I have ordered the USB to TTL adapter and it'll show up in a couple of days.

ww123td, 2023-04-07:
I have set up my serial adapter, please let me know what steps to follow next.

jfedor2, 2023-04-07:
OK, connect pin 0 on the "B" Pico to RX on the serial adapter and one of the Pico GND pins to GND on the adapter.

And try this firmware (included both A and B, but it's really the B side that's of interest to us right now.)

[debug_firmware.zip](https://github.com/jfedor2/hid-remapper/files/11178493/debug_firmware.zip)

Get some serial terminal software on the PC and set the baud rate to 921600. Then plug the trackball in. You should be seeing some output similar to this (and a lot more):

```
[0:] USBH DEVICE ATTACH
hcd_port_reset
hcd_port_connect_status
Full Speed
```

No guarantee this will help us make it work, but who knows.

ww123td, 2023-04-07:
Here's the debug output
[Huge Debug Output.txt](https://github.com/jfedor2/hid-remapper/files/11178860/Huge.Debug.Output.txt)

I noticed that if I press the buttons on the mouse, sometimes I would see this:
`
[0:0:0] USBH DEVICE REMOVED
  Address = 1
HID close
tuh_hid_umount_cb 1 0
HUB close
hcd_device_close 1
`
and then the same initialization sequence recorded in the file would commence.

jfedor2, 2023-04-07:
I was maybe expecting some errors during enumeration, but that seems to go though OK.

Is there any further output when you move the ball or press the buttons (when it doesn't cause a disconnect)?

jfedor2, 2023-04-08:
Yeah, you can see some errors in those. But it would be good to catch the moment when it first goes wrong. Because the first log looks normal. Does it send nothing on the serial port after that when you interact with the trackball?

One things that's perhaps unusual is that wMaxPacketSize is 8 bytes even though it's a Full Speed device so one could expect it to be 64, but I think that's legal and I don't see any obvious place where it would cause issues.

ww123td, 2023-04-09:
Yeah, it didn't send anything after that, I had to power cycle the Pico to
get it to send any input to the computer.

I had been unable to recreate the issue since, so there's nothing new to
look at from my end.

MetalMan1, 2026-11-06:
Hello, I am finding the same issue as @jmdesprez with an Elecom Huge wired. My HID Remapper is Custom V1 board manufactured by JLCPCB, firmware is latest r2026-05-25 remapper_board.uf2. I tested with my keyboard and everything works beautifully, I can see keyboard activity in the Monitor tab. But with Elecom Huge, there is no activity from the mouse and no activity in Monitor tab. When plugging it in I see @jmdesprez same LED blinking behavior.

I provided 2 screenshots. #1 is showing all the devices which enumerate in Device Manager (Windows 11) when Elecom Huge is plugged in. #2 is the hardware ID for the specific Elecom Huge entry.

Do you have any ideas of something I could try? I really hope to use HID Remapper with my Elecom Huge (at home and work) as I would like to ditch their software at home and I can't install it at work.

[image]

[image]