jfedor2/hid-remapper #57: Compatibility with kmk? (and many questions)

jacostag, 2023-04-14:
Hello, before anything thank you for this great project, it does not have all the attention that should. Is a super easy and affordable way to have a different layout without messing with the OS. The only other alternative is the hasu usb converter and is way more difficult to assemble or expensive to buy.
This is basically soldering 4 cables, dropping a file and opening a web!

I have not found much info on what is actually running under, I know this project has gain some attention for mouse "speed", but I know there is a lot of people that wants an easy and affordable way to have a programmable keyboard.

Can this project run kmk/qmk? or is there a compatibility to use the same json format to share layouts?

Can this project be used on https://www.adafruit.com/product/5723 ? is a rp2040 with an usb host a support, so, maybe not even soldering required.

I have a keyboard with an usb host (you can connect an extra peripheral like a mouse) and it does not recognize any key, is possible to test what the pi is seeing?

You should have a discord or something that allows people to discuss your projects ;)

Lastly, sorry to write so many things in here, but I did not find any other community to ask you.

ATMakersBill, 2023-04-14:
I have some of those USB Host feathers on order - when I get them I intend to try this project on it (as well as their IntelliKeys adapter) and will report back.  I don't see why it wouldn't

jacostag, 2023-04-17:
> I have some of those USB Host feathers on order - when I get them I intend to try this project on it (as well as their IntelliKeys adapter) and will report back. I don't see why it wouldn't

Thank you, yeah it should work, it is the same rp but not sure if pins are different, maybe the fw needs one or two changes to make it work. Anyhow, please let us know your findings

ATMakersBill, 2023-04-25:
I have tested the latest pre-release r2023-04-24 thanks to a heads up from @jfedor2 

It works like a charm on the new Feather boards!

Just download the remapper_feather.uf2 file and load it on the board.  Configs & runs as expected.

https://github.com/jfedor2/hid-remapper/releases/tag/r2023-04-24

jacostag, 2023-04-25:
@ATMakersBill thank you, now is even easier to use this project :)
Is there a place to share layouts? @jfedor2 or have you thought on the ability to use tmk,qmk or kmk? Thank you for this nice project

jfedor2, 2023-04-25:
@jacostag Not sure what you mean by use or run QMK. If you mean is it possible to make a converter that would take QMK layout JSON and produce equivalent HID Remapper configuration JSON then possibly, but aren't QMK layouts dependent on the physical matrix of the device? I only used QMK for DIY split ergo keyboards so maybe I'm missing something here.

jfedor2, 2023-04-25:
When you connect your keyboard to HID Remapper and press some keys, does the LED blink on the Pico? Are you using the single or dual Pico version? It's possible the keyboard has a USB hub inside that's not compatible for some reason. Though many hubs do work.

jacostag, 2023-04-26:
@jfedor2 no led is turning on when I press any key
For the usb to usb is basically something like what you created, only for keyboard, but is more complex/expensive to have
Original using tmk
https://geekhack.org/index.php?topic=69169.0

This is the port to use qmk usb to usb
https://github.com/qmk/qmk_firmware/tree/master/keyboards/converter/usb_usb

But I have not seen anyone using it yet. Thats why I was thinking if you managed to use qmk or kmk on your device it will be cheap and easy to have a "programable keyboard" just using your device (and also using qmk/kmk there are ton of community support)

dglaude, 2023-04-26:
Please for tester on the Feather RP2040 USB HOST from Adafruit, have a look at #60

jfedor2, 2023-04-26:
> This is the port to use qmk usb to usb https://github.com/qmk/qmk_firmware/tree/master/keyboards/converter/usb_usb
>
> But I have not seen anyone using it yet. Thats why I was thinking if you managed [...]

Oh, I see what you mean now. With some effort someone could probably make it run on a Pico (or two), but that question is better directed at people who made that project. :)

jfedor2, 2023-04-26:
Regarding your keyboard, the only thing I can recommend is trying the dual Pico version, but even then there's no guarantee it will work.

jacostag, 2023-04-28:
Thank you for all the answers!

closed 2023-04-28 by jacostag