jfedor2/hid-remapper #58: Improve KVM switch compatibility

braxlan, 2023-04-20:
Hi,

first of all: great project!

Unfortunately it does not work on any of my KVM switches.

I guess the reason is that it is running in full speed mode on the device side, which the HID emulation of the KVM switches do not like. Would it be possible to introduce low speed mode on the device side as an option? This would be greatly appreciated! I took a quick look at the code but could not enable low speed mode by myself.

Another small problem could be the fact that the virtual keyboard and mouse do not identify themselves with the right protocol ("none" instead of "keyboard" or "mouse"). This is a quick fix, I can offer a pull request if you like. Obviously, I cannot test if this makes any difference for the KVM switches, because it is not running in low speed mode right now.

Best regards,
braxlan

jfedor2, 2023-04-20:
I doubt it's about Low Speed/Full Speed. Your KVM switches probably expect the connected devices to operate in "boot protocol" mode, which is essentially a fixed report descriptor plus the identifying bits you mention. Supporting that could be beneficial in general, because it would potentially make it work with some BIOS-es that also expect boot protocol. But it might require separating mouse and keyboard to their own interfaces. That's not impossible, we already have a separate interface for configuration.

braxlan, 2023-04-27:
Thank you for the hint, I am not very deep into USB. How can I find out if a device supports boot protocol mode, for example using `lsusb`?

jfedor2, 2023-04-28:
In lsusb output it will say something like this in the interface descriptor if the device can work as a boot protocol keyboard:
```
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
```

Most normal keyboards will have that. Often a device has two or three interfaces, especially if it's a wireless dongle that supports both keyboards and mice, then it will have one interface as above and another with bInterfaceProtocol=2 (Mouse).

braxlan, 2023-04-30:
After your helpful explanation I diagnosed all devices which I tried to connect to the KVM switch console. All of them support boot protocol mode, but those who don't work have one thing in common: They run in Full Speed. Only devices with Low Speed mode are working.

Also the HID emulation of the KVM switches itself presents Low Speed devices, and I can connect each KVM switch to the other in an arbitrary order with working HID emulation. The KVM switches are made by two different manufacturers. One of them presents one device with two Low Speed HID subdevices (composite), and the other one presents a hub with two Low Speed HID devices behind (compound).

jfedor2, 2023-04-30:
I see. Anyway, I don't think it's possible to make the RP2040 operate in Low Speed mode as a device.

braxlan, 2023-05-04:
Is it possible to divide the USB clock on the device side by 8?

braxlan, 2023-05-04:
You read my mind! ;-)

From what I understood it is possible to configure the clock dividers, so maybe there is a chance for my plan to work. I'm not sure about the quirks it might bring to the overall timing concept of your project. It also might be a bit off-scope to make Low Speed work, so I would totally understand if you don't give this any priority.

braxlan, 2023-07-21:
I just saw that the USB input/output clocks are now synchronized. Would it be possible to introduce a clock divisor of 8 here?

jfedor2, 2023-07-21:
Not really. Not in the way you're interested in anyway. We're not doing anything to the device side clock, we're just synchronizing the bit-banged host side clock _to_ the native one.

braxlan, 2023-07-21:
Ah, so we would still have to divide the device side clock by 8 and multiply by 8 again on the bit-banged side. I would love to test this. I looked up the registers and (hopefully) found the right one, but did not find anything about how to write to the registers.

braxlan, 2023-10-04:
Closed as irrelevant. Will open another issue explicitly for Boot Protocol Mode.

closed 2023-10-04 by braxlan