jfedor2/hid-remapper #60: feather_rp2040_usb_host need to have GPIO18 enabled to USB to work

dglaude, 2023-04-26:
The USB 5V on the Feather is controlled by RP2040 GPIO18.
That need to be set to ouput and "True" to have the Green LED next to the USB turn on and the external keyboard/mouse be powered.
Else the behaviour is random, and my board does not behave like the one of ATmaker that is working on a fix.

ATMakersBill, 2023-04-26:
I've got a fix to this (its' straightforward) and it's confirmed - I can turn on and off the USB at will.  Might also be a good feature to expose via the web interface (handy to turn off/on the device). 

I'm setting up a fork so I can do a pull request with this fix, but for anyone waiting the fix is easy:
Add 
`#include <pico/stdlib.h>`
to the top of main.cc (not inside the extern "C" block)
then add this just before the main loop
```
    /* On the Feather Host board, the 5V boost converter needs to be turned on
       If not, the pin is floating and it may/may not be enabled */
#ifdef FEATHER_HOST_BOARD
#define POWER_5V_PIN 18
    gpio_init(POWER_5V_PIN);
    gpio_set_dir(POWER_5V_PIN, GPIO_OUT);
    gpio_put(POWER_5V_PIN, 1);
#endif
```

jfedor2, 2023-04-26:
Ugh. There's supposed to be a pull-up resistor on that pin (at least it's on the schematic).

ATMakersBill, 2023-04-26:
Well, it's an easy fix - I have a fork setup - I was going to add the log() stuff to it but I can do this first if you want a pull request.  If not you can just copy/paste.

It might be that the pull-up's not adequate or it's a defect in the board.  Not a bad idea to set the pin anyway

ATMakersBill, 2023-04-26:
Ok - I'll leave this to you then - David's got a custom build and is making progress

dglaude, 2023-04-26:
> Yeah, it's no problem, driving that pin is even already present in TinyUSB (https://github.com/hathach/tinyusb/blame/master/hw/bsp/rp2040/family.c#L128) so it's just a matter of adding a define. I just thought it wasn't necessary.

Sorry, I cannot explain the why...
The board of @ATMakersBill was working without this fix.
My board does not power up the 5V without this patch.

First I was thinking my board was dead on arrival.
So I booted into Circuit Python where I was able to turn on the power with GPIO18.
My board was OK but not behaving exactly like his board.

@ATMakersBill tested on his board that he could turn off the power on his board, and then reversed to turn on the power. I tested his patched UF2 and it works for me (and still works for him).

For me, this change is a must have, else I cannot even use the board.

jfedor2, 2023-04-26:
Of course, it's no problem. Please try this build if you can, it should contain the same fix:

https://github.com/jfedor2/hid-remapper/releases/download/r2023-04-26/remapper_feather.uf2

dglaude, 2023-04-26:
It works perfectly.
( also it is really useful to be able to jump to UF2 mode directly from https://www.jfedor.org/hid-remapper-config/ )

jfedor2, 2023-04-26:
Awesome, thank you!

closed 2023-05-26 by jfedor2