jfedor2/hid-remapper #64: GUI key "behaves differently" with and without hid-remapper

dov, 2023-05-02:
What I did:

1. I'm running a Linux system under X11
2. I have two keyboards connected to the computer. One through hid-remapper, and one connected directly.
3. My window manager is catching GUI key combinations, called Super_L under X11, and has assigned certain window manager actions to it.
4. When pressing the Super_L+T on my fixed keyboard, the window title is toggled as expected.
5. When pressing the Super_L+T on through hid-remapper, it is not recognized by  my window manager.

I have tried  to compare the differences between the two events through `xev`, but the keyboard events from both  keyboards  appear identical.

I have verified that if I connect the additional keyboard directly to the computer, without the remapper device, it works as expected.

jfedor2, 2023-05-02:
Can you dump what HID Remapper is sending with something like:
```
usbhid-dump -d cafe:baf2 -es
```
?

dov, 2023-05-02:
Ok, here is the output from `usbhid-dump`. From the same run I pressed the sequence "Super_L press", "t press", "t release", "Super_L release". First I did it on my USB keyboard, and then I did the same sequence on my hid-mapper keyboard:

```
$ usbhid-dump -es                                                         1 ↵
Starting dumping interrupt transfer stream
with 1 minute timeout.

001:004:000:STREAM             1683057331.560461    ← Super_L press on USB keyboard
 08 00 00 00 00 00 00 00

001:004:000:STREAM             1683057331.824471    ← t press
 08 00 0E 00 00 00 00 00

001:004:000:STREAM             1683057331.928430    ← t release
 08 00 00 00 00 00 00 00

001:004:000:STREAM             1683057332.112426   ← Super_L release
 00 00 00 00 00 00 00 00

001:038:000:STREAM             1683057338.264283    ← Super_L press on hid-mapper keyboard
 02 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00

001:038:000:STREAM             1683057338.486262    ← t press 
 02 08 00 00 08 00 00 00 00 00 00 00 00 00 00 00
 00

001:038:000:STREAM             1683057338.546258    ← t release
 02 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00

001:038:000:STREAM             1683057338.652227    ← Super_L release
 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00

[1]    36440 terminated  usbhid-dump -es

```

(Note that I'm using the Dvorak keyboard layout, so "t" is on the physical key "k")

jfedor2, 2023-05-02:
HID Remapper is sending "T". I assume your PC is then doing the Dvorak remapping so it ends up being something else. But why would it be sending "T"? I understand you're pressing the "K" key on your keyboard (in terms of the standard layout). What configuration do you have on the HID Remapper?

dov, 2023-05-03:
Thanks. X11 is peculiar in that if you remap one keyboard through `xkbcomp` and then connect a second keyboard, then the second keyboard is not remapped automatically. So I was in the situation where my USB keyboard was under X11 remapping, but my hdi remapper keyboard was not remapped by X11. On the other hand, I had a full Dvorak remapping active in HID-Remapper. I'll redo the test without any remapping and will check the result.

In any case, not even I can really find a use case, where I will need the Super working as a modifier, where I can't do the remapping on the OS level... :sweat_smile: