jfedor2/hid-remapper #65: RE: Limitations on passing through Macros & Text Strings

24-7media, 2023-05-03:
Hi, Great work on the HID Remapper to date…

This is looking to be a great plug and play solution for Non-native Bluetooth/BLE PC’s (e.g., Windows 7 etc.) once the devices are paired.

My uses are somewhat a bit simpler than most of the other users and my feedback is a little limited in not being a programmer/engineer myself.

I’ve been testing the HID Remapper under Windows OS (Win 7 & 10) with Adafruits nRF5280 Feather & XIAO nRF52840 boards acting as the Bluetooth Receiver/Remapper, I have these successfully paired with a Adafruit nRF5280 Feather setup as a BLE HID Macro device for sending macros/keystrokes combinations etc.

So far, I have all the inputs Unmapped and set to passthrough on all layers, with all keystrokes/macros, mouse clicks & consumer control codes being passed through.

The only problem I am finding is that there appears to be a limitation on passing through longer macros/text strings etc. These appear to timeout and get clipped or on some occasions loop and repeat on the last keystroke etc. Sometimes it can be erratic in that some characters are left out of the string.

I can use a combination of Modifiers, Delay/Sleep tags with keystrokes to build and input the macros, but I’m finding a limitation with ability to passthrough longer strings, e.g., Web Addresses etc.

I don’t find any issues with the above when pairing directly via BLE to the PC, just seems to be something I am coming across when paired and passing macros through the HID Remapper.

Also, I’m finding that the ‘Forget all devices’ option under the Web Config Tool, isn’t unpairing the connected nRF52840 Feather from the BLE Receiver/Remapper.

Question, would there be any possibility in the future to be able to rename the ‘HID Remapper Bluetooth’ device? Just so it could be renamed to something more descriptive to the device it is paired too etc.

Appreciate your efforts so far, it’s great having a BLE HID solution for older non-BLE hardware etc.

24-7media, 2023-06-05:
Hi Thanks for getting back to me and apologies for the delayed reply.

I have updated the seed xiao nRF52840 to your latest UF2 (r2023-05-26), the software running on the other nRF52840 feather is CircuitPython 7.3.3 with the adafruit ble & hid libraries.

I have also tested CircuitPython 8.1.0 with the same results.

jfedor2, 2024-09-25:
I know it's been a while, but I made a test firmware build that might address this issue. It's here (the "artifact" link at the bottom):

https://github.com/jfedor2/hid-remapper/actions/runs/11039276703