jfedor2/hid-remapper #67: FR: Please add more stack expression (SWAP, ROT)

dglaude, 2023-05-04:
Hello,

I am playing with the expressions to try to implement some Tremor filter (see #66) so I try to see if the mathematical function I might need are present or not.

But before dealing with mathematical function, I think it would help to have a few more stack management functions. Right now we seems to only have `dup` but I think it would be useful to have a few more.

Notice that I am not blocked yet by any missing stack function, but it could help or simplify some expression in the future.

Good ideas of missing stack functions can be taken from Chapter 2.1 Stack manipulation starting on page 4 from this: https://literature.hpcalc.org/community/prog-userrpl.pdf

I think the bare minimum would be:
* SWAP(x,y) exchange the two topmost position of the stack
* PICK(x,stack) copy the Xth element from the stack and put that on top
* ROT(x,y,z) rotate the 3 first elements so that the 3rd move to the top
* OVER(x,y) make a copy of y on top of the stack

Of course many other could be interesting, but since we have no test/conditional function, DROP and DEPTH seems not very useful right now.

I have not looked at the code yet, but I might have enough skills to try to PR some stuffs if you prefer.

Let's first have a conversation on what you think is missing/important/acceptable.

Regards

jfedor2, 2023-05-06:
I'm not against adding some more operations, but let's see some use cases. What expression did you want to write that was possible/easier/more elegant with these operations?

dglaude, 2023-05-06:
Right now I have work around the stack limitation by doing the computation I needed in the order I needed them on the stack for later operations...

But I had more trouble with missing operations:
* minus (it is missing, so you can add a negative number or multiply by -1 and add)
* divide (I had to divide by a constant, so I did multiply by the inverse... but that would not be possible with a variable value)