jfedor2/hid-remapper #68: Question: Why use the 2 pico variant vs the 1 pico variant?

apastuszak, 2023-05-09:
I want to build one of these. But I am not sure why you would use the 2 pico variant instead of the 1 pico variant.

jfedor2, 2023-05-09:
"They have the same functionality, but the dual Pico version has better device compatibility - most input devices work with either, but some will only work with the dual Pico version."

apastuszak, 2023-05-09:
Does anyone sell these prebuilt?Sent from my iPhone

jfedor2, 2023-05-09:
Adafruit does!

https://www.adafruit.com/product/5723

schoerg, 2023-05-09:
>Out of stock

closed 2023-05-26 by jfedor2

alexanderwiller, 2023-07-05:
Hello,

I am considering using the RP2040 for an application that is basically what this hid-remapper does, but in a more complex setup with multiple RP2040 controllers that should be each host and device.

Now I'm wondering about the details of compatibility limitations with the PIO-based USB implementation, as it would be highly advantageous to not have to use two RP2040 controllers per host-device pair.

I've read through existing issues, the HW docs and the Pico-PIO-USB repo as well - but could not figure out where the compatibility limitations come from.

Can anybody provide some more information on this topic? Are the limitations hardware-wise or is it more that the Pico-PIO-USB library or the TinyUSB library has limitations, but they could probably be solved in the future?

Does it make any difference to use the HW USB port for upstream connectivity and the emulated USB port for downstream connectivity versus the other way round?