jfedor2/hid-remapper #69: feature request: hard reset

itsnoteasy, 2023-05-10:
I have some hardware, both logitech mouses, g102 and g502, also a teensy 2.0 based keyboard, that occasionally need to be unplugged and re inserted to wake them up. I haven't used any of them with the hid remapper yet, but i was looking at a video on youtube of how usb speed is negotiated by enabling the pullup resistors one at a time on d+ and d-.  link here https://hackaday.com/2023/05/09/minimal-usb-device-connects-with-just-a-couple-of-resistors/#comments

anyway, i figure you could add a watchdog timer feature that detects when a device stops responding and if so, disables the pullups for a few seconds so it appears that the device has been unplugged. i'm not sure if that would work to wake them up, but it seems like an easy thing to do so what do you think.

ATMakersBill, 2023-05-10:
If you're using the Adafruit Feather USB Host, you can also turn off the 5V regulator via a pin (18?) and that would completely reset the device

itsnoteasy, 2023-05-11:
good to know, but that device would be rare both due to limited availability and increased price compared to the alternative. I'm  not sure is the proposal a good idea even. not everyone wants to run buggy hardware that could die at any time, but a device that only dies at system boot is usable.

closed 2023-05-23 by itsnoteasy