jfedor2/hid-remapper #70: Crazy behaviour with rollover/ghosting

columna1, 2023-05-10:
Using this HID remapper I sometimes get really weird key repeats and such when typing. I have done some research and have come to the conclusion that this likely has to do with ghosting/rollover sending all 0x01 bytes in the keycode report when there is an error state. This likely causes prev_input_state to be overwritten with all 0x01 or something similar making hid-remapper think the keys are no longer pressed. Then, when the keyboard comes out of error state and sends the rest of the keys currently pressed hid-remapper thinks that all of those keys are pressed sends them all again a second time.
this results in things like in colemak the "oul" keys when pressed on a cheap hp keyboard gives an error state. (; I and U keys in qwerty) leading to things like typing "could" resulting in "cououllud" or something similar. I would assume we want to ignore packets that have that error state or at least prevent updating the keyboard state when one is received.

jfedor2, 2023-05-11:
Good catch!

columna1, 2023-11-22:
I just want to clarify, on some keyboards this can occur when pressing as little as 3 keys at once, it's not just a 6kRO bug. That's why it was an issue with my cheap company provided keyboard at work. I was hitting this when typing normally.

jfedor2, 2023-11-22:
I haven't forgotten about this!

jfedor2, 2023-11-23:
If you still have access to that keyboard and are up for some testing, you can try a test build from here (the artifact link at the bottom):

https://github.com/jfedor2/hid-remapper/actions/runs/6973886250

I'm not sure if it's necessarily the solution I want to go with, but it's one possible solution.

columna1, 2023-11-25:
Sorry for the delay. I've managed to test the firmware you provided and it works great.

jfedor2, 2024-01-10:
The latest official release also contains a (different) fix for this!