jfedor2/hid-remapper #71: Support question: Is there a way to implement a clicker

dglaude, 2023-05-22:
Maybe this is a feature request, but maybe it is already possible?

Just like there is a mouse jiggle made with expression, I could like to implement (with expression?) a clicker.
I would need a time based (left) click of the mouse.
Right now, I don't really see how this can be done, as the only usage of expression I have seen have been for mouse mouvement.
But maybe I am missing some of the power just by ignorance or by not reading carefully the documentation.

Regards

jfedor2, 2023-05-22:
It's like any other expression, it just has to generate 0 or 1. Then create a mapping with "Expression n" as input and "Left button" as output. If you want to check if a button is pressed, you can use usage 0x00090001 for left button etc.

closed 2023-05-26 by jfedor2

dglaude, 2023-05-26:
For the record, I was successful at implementing a clicker (and a left/right clicker that oscillate between two positions) but it is a bit tricky. The issue is getting the right timing, like if you just want to send one click or I guess more complicate if you want to do a double click.
Maybe I need to double check the frequency at witch the expression is evaluated.

jfedor2, 2023-05-26:
Feel free to post your expressions along with the intended results.

Frequency shouldn't matter, at least not until we have something like registers that can keep state. The expression answers the question "at time X, is the button pressed or not". Depending on what inputs come from connected devices, it might be evaluated more or less often, but again, it shouldn't matter.