jfedor2/hid-remapper #74: FR: add a comparator mode for v scroll with h scroll

itsnoteasy, 2023-05-30:
I often scroll down pages and images with a trackball, but when i have horizontal scroll on the same layer then the page slides left or right. although separate layers are usable, it would be handier if hid remapper was able to detect whether i was moving mostly horizontally or vertically , and ignore the lesser inputs.

jfedor2, 2023-05-31:
You can try to do it with expressions. For example:

Expression 1:
`0x00010030 input_state dup abs 0x00010031 input_state abs gt mul`

Expression 2:
`0x00010031 input_state dup abs 0x00010030 input_state abs gt mul`

Then put those expressions as inputs in place of "Cursor X" and "Cursor Y" in your mappings.

Might not work perfectly depending on the specifics of your trackball, but it's a start.

To do it properly we'd need some way of averaging inputs over time, which might be useful for other things as well so it's something I'm thinking about.

itsnoteasy, 2023-05-31:
It didn't work for me but thanks anyway. those expressions are all greek to me, is there a guide you could point me towards

jfedor2, 2023-05-31:
They're actually Polish, not Greek. Reverse Polish, to be precise.

And I tried to describe them here:

https://github.com/jfedor2/hid-remapper/blob/master/EXPRESSIONS.md

dglaude, 2023-06-01:
> 0x00010030 input_state dup abs 0x00010031 input_state abs gt mul

It is not really Greek, but maybe Polish... as in "Reverse Polish Lisp" or other stack base calculator (also like Forth).

So that would do (I could have some details wrong):
1) Take current X input value
2) Make a copy of that on the stack
3) Take the absolute value of that value from (2)
4) Take the current Y input value
5) Take the absolue value of value from (4)
6) Compare if (5) is greater than (3) => absolute(X)>absolute(Y) => put 0 if false and 1 if true
7) Multiply the value of (1) by the value of (6)

So all in all, if the absolute X movement is bigger than the absolute Y movement, then use that X movement.
But if the absolute Y movement is bigger, then use zero.
And that is your new X movement.

By doing that on X and similar on Y, your mouse will only work horizontal or vertical... and removing any non orthogonal mouvement.

Except for understanding the stack and reverse order of doing math (such as "1 2 add" is the equivalent of "1+2") you have to understand the trick of doing a test and multiplying by zero (the false value) to use the result of the test.

I have encountered limitation and there are things where I had to use convoluted formula to get what I wanted to do, but it is very capable and you can do impressive things.