jfedor2/hid-remapper #75: Monitor tab doesn't recognize extra buttons

whiskthecat, 2023-05-30:
Hello, I am using Elecom Huge and the monitor tab doesn't respond when I click Fn1 Fn2 or Fn3.

jfedor2, 2023-05-30:
Do the other buttons show up in the monitor tab?

Is it wired or wireless Huge?

whiskthecat, 2023-05-31:
It's wireless huge. Yes the normal mouse buttons and the cursor show up in the monitor tab. I tried running wireshark USB capture to figure it out but I'm not making any sense out of it and there are tons of hex codes on the input selection page.

jfedor2, 2023-05-31:
It seems you have a new variant of the Huge. We already have some special handling for product ID 0x010d, but yours is 0x011c.

I found some related patches in the Linux kernel, but it's not clear to me if the new variant has the same report descriptor (it's definitely broken in a similar way though).

Can you try to dump the report descriptor with Wireshark again? Just do the same thing you did previously, but unplug and re-plug the device while the capture is running.

(I know it might seem that this is what the Monitor tab was supposed to save us from, but sadly it doesn't, at least in its current version. If the report descriptor is broken then we still can't access the inputs. Something to consider in the future.)

whiskthecat, 2023-06-01:
[hugeplug.zip](https://github.com/jfedor2/hid-remapper/files/11625303/hugeplug.zip)

jfedor2, 2023-06-01:
Perfect, thanks.

The descriptor is indeed different, but still broken.

Please try this firmware build and let me know if it helps.

[elecom_huge_011c.zip](https://github.com/jfedor2/hid-remapper/files/11627534/elecom_huge_011c.zip)

whiskthecat, 2023-06-01:
That worked, you're the man!
[image]

jfedor2, 2023-06-01:
Great! I will include the fix in the next release.

jfedor2, 2023-07-07:
This is now in the latest release.

closed 2023-07-07 by jfedor2