jfedor2/hid-remapper #76: Inconsistent scrolling speed (Pico + Elecom Huge)

joseph423, 2023-06-03:
I am using Pico + Elecom Huge.

I created a new layer dedicated for trackball scrolling (Mapping "Cursor Y" to "V scroll" with scaling set to 1).

However, sometimes the scrolling speed turns extremely fast. So I set the scaling 0.01, which the scrolling speed feels so great.

And then somehow the speed turned real slow (happens sometimes in the middle of using it or unplug&plug again the pico), so I set the scaling to 1 again......

** I do not find a very consistent pattern that trigger this, anyone found the same?

jfedor2, 2023-06-03:
What operating system are you using?

Might be some difference of opinion between the PC and HID Remapper as to whether we are in high-resolution scroll mode or not. Not sure what could be causing it.

joseph423, 2023-06-03:
I am on Windows 11.

There was once I use it on MacOS, and if I remember that correctly, the same happened.

> What operating system are you using?
> 
> Might be some difference of opinion between the PC and HID Remapper as to whether we are in high-resolution scroll mode or not. Not sure what could be causing it.

joseph423, 2023-06-03:
The same happened after I create this post LOL. (Now I set the scaling back to 1 instead of 0.01)

I wonder is there a way to debug it.

The first thing I want to make sure, is that the PICO is outputing the correct amount/value of V scroll while I keep the scale as 1.

Sometimes, the scale 1 works normal, and sometimes it becomes extremely fast. Unplug and plug again resolve it(Sometimes......)

222Phoenix, 2024-04-22:
I've always had this issue too. Scrolling (both trackball and scroll wheel) work great, then after some period of time, scrolling becomes usably slow but cursor still works normally. If I unplug and replug then scrolling goes back to normal. Device is directly connected, not through KVM, and had this issue on win 10 and 11.

jfedor2, 2024-04-22:
Interesting. What is the approximate time after which this happens? Does you computer go to sleep by any chance?

222Phoenix, 2024-04-22:
I think the time period was several days, maybe a week. I'll make a note of it for next time. Yes, usually the computer goes to sleep nightly instead of shutting down. Maybe twice a day sometimes. Is there an issue with sleep?

jfedor2, 2024-04-22:
I don't know, I'm trying to guess the cause. Just time by itself could be it if some counter wraps, but the thing with scroll is that HID Remapper (and any mouse) starts in low-res scroll mode by default and the OS has to issue a command to switch it to high-res mode. But sometimes the device needs to switch back to low-res scroll by itself if it thinks there's been a reboot. Maybe what's happening is HID Remapper switches back to low-res mode, but the OS still thinks it's in high-res mode and doesn't re-issue the switch command. But that's just speculation.

joseph423, 2024-04-23:
@222Phoenix @jfedor2 
I have some findings by the end of last year, these info may help.

Will keep you guys posted later to day.

222Phoenix, 2024-05-01:
Thanks @jfedor2! I will do some testing and see what I can find out.

mastoca, 2024-05-08:
@222Phoenix how did your testing go? I've seen the same issue with a Kensington trackball on MacOS where scroll will jump after a certain distance.

222Phoenix, 2024-05-09:
> @222Phoenix how did your testing go? I've seen the same issue with a Kensington trackball on MacOS where scroll will jump after a certain distance.

So far so good with the above firmware, but I haven't been able to do much testing. I'm having some computer issues, might have to reinstall Win11. I think my issue is different than yours though. I didn't have any jumping, just extremely slow scrolling, presumably after several sleeps/wakes on Windows.

222Phoenix, 2024-05-20:
> @222Phoenix I made a test build of the firmware that never resets the high-resolution scroll multiplier, give that a go if you can. It's not the easiest thing to troubleshoot when it only happens every few days, but maybe that's it.
> 
> https://github.com/jfedor2/hid-remapper/actions/runs/8914743419
> 
> ("artifact" link at the bottom)

It took a while, but I experienced the same super slow scrolling issue on Win11 with the test firmware. I think it takes several sleeps/wakes for it to happen.