jfedor2/hid-remapper #77: Feature Request: Meet Mute Button mapping

davewongillies, 2023-06-09:
I don't know if this is feasible but it'd be great to be able to map the same mute button as in your https://github.com/jfedor2/meet-mute-button project

dglaude, 2023-07-05:
The shortcut to Mute seems to be CTRL+D: https://support.google.com/meet/answer/9298571?hl=en
So to have the same effect you could:
* Take a Feather RP2040 USB Host board
* Use the boot button as the button to press (that is GPIO 7)
* Map the GPIO 7 to Macro 1
* Define Macro 1 to be Left (or Right) Control + D

And the result is a physical mute button for Google Meet.

[image]

[image]

You could also use another GPIO and solder those pin to a big button.

I hope it help (I tested on my computer).

PS: Here is my json file exported:
```{
    "version": 7,
    "unmapped_passthrough_layers": [
        0,
        1,
        2,
        3
    ],
    "partial_scroll_timeout": 1000000,
    "tap_hold_threshold": 200000,
    "gpio_debounce_time_ms": 5,
    "interval_override": 0,
    "mappings": [
        {
            "target_usage": "0xfff20001",
            "source_usage": "0xfff40007",
            "scaling": 1000,
            "layers": [
                0
            ],
            "sticky": false,
            "tap": false,
            "hold": false
        }
    ],
    "macros": [
        [
            [
                "0x000700e0",
                "0x00070007"
            ]
        ],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        []
    ],
    "expressions": [
        "",
        "",
        "",
        "",
        "",
        "",
        "",
        ""
    ]
}```

jfedor2, 2023-07-05:
That's different. Ctrl-D will only work when the window with the video conference has focus. The linked mute button project works even if another application is active.

dglaude, 2023-07-06:
Oh that explain a lot.

I was trying to read the code of your other project, and I could not locate the CTRL+D and I was wondering how it all work... so I did it my way.

The other thing that I was interested in is the color ring and that I did not have a way within hid-remapper to control the NeoPixel build into the Feather USB Host hardware. That could be an interesting general purpose feature to indicate the layer or activity. And I was also impress by one of your tweet that show a LCD attached to a kind of prototype. Since there is an I2C Stemma QT port on the Feather, and numerous LCD by Adafruit or other, it was also a sexy feature to have.

closed 2026-08-30 by davewongillies