jfedor2/hid-remapper #78: Only one device inputs picked up when connected to USB hub

jarrodharvey, 2023-06-29:
I'm using an adafruit feather rp2040. The remapper can pick up the presence of one device but not two; if I plug in a hub with two devices attached only one will register. This applies for both software (the monitoring tab) and hardware (the light on the rp2040 that flashes when an input is passing through it). 

The device that "connects" seems to be random, but if I connect a hub with one device plugged in then that device will be detected and any subsequent devices plugged into the hub will be ignored. 

The hubs that I have tested with so far have not had an external power supply, although I have ordered a hub with an external power supply off Amazon to see if power draw is an issue and will find out once it arrives.

jarrodharvey, 2023-07-24:
Update: I have attempted to use a custom board ordered from JLCBCB connected to a hub with an external power supply. Still no luck, unfortunately.

Those who have had success with hubs: how did you do it?

jfedor2, 2023-07-24:
In my experience some hubs work and others don't, there's no trick to it.

I'm not sure if I've seen this behavior where only one connected device works though, usually when it doesn't work, it just doesn't work at all.

jarrodharvey, 2023-07-25:
Thanks jfedor2. Do you have an example of a working hub that you use?

jfedor2, 2023-08-03:
Not sure how much this helps because I got these hubs a long time ago and they might be hard to track down, but these two work:

[image]

[image]

They show up as:

`058f:9254 Alcor Micro Corp. Hub`

and:

`1a40:0101 Terminus Technology Inc. Hub`

I also have a UGREEN hub that works partially: it only works if I plug devices into the hub when the hub is already plugged into the remapper.

miwagner1, 2024-04-01:
Insignia NS-PH3A4AT USB 3.0 4-port Travel Hub does not work from Best Buy. Time to keep buying until one works.

  Product ID:	0x0817
  Vendor ID:	0x2109  (VIA Labs, Inc.)
  Version:	90.13
  Serial Number:	000000000
  Speed:	Up to 5 Gb/s
  Manufacturer:	VIA Labs, Inc.

jarrodharvey, 2024-04-01:
I should provide an update of my own - I still haven't found a hub that works. 

It's not strictly a usb hub, but there is a device called the Microsoft Accessory Hub that is compatible with this, though. You are extremely limited by what you can plug into it - you can only plug in 3.5mm devices or products designed explicitly by Microsoft for the purpose of connecting to it. This information is somewhat valuable to me. While it might not be useful to many people, I'm putting it out there just in case it is.

miwagner1, 2024-04-02:
> I should provide an update of my own - I still haven't found a hub that works.
>
> It's not strictly a usb hub, but there is a device called the Microsoft Accessory Hub that is compatible with this, [...]

It could be that it’s hating on something being 3.0 or too much power draw (memories when I had to use a usb 2.0 hub…. Long thrown away 😅 when I managed to avoided the power consumption message on my old iPhone 4).

This hub information may help a lot and as that accessory is USB 2.0 full speed only (and now I kinda want one… down the rabbit hole I go). I’m going to pick up some more very basic usb 2.0 hubs tomorrow for testing and dig around to see if I can’t find my old lightning to usb adapter to test these hubs at the stores with to see if they give me a power draw message. I’m going to buy the ones that don’t and see if they work. (I still have that old fiio mini usb battery dac to test with)

Unfortunately I am not sure how to debug this yet as I am rather new to the microcontroller scene as I bought this for my kensington expert mouse after I had early stages of carpal tunnel (burning wrist before the next stage of numbness). The Kingston software UI can’t see my trackball when my work dock is plugged in (but the engine still works 🤯)which lead me here.

Thank you soo much for making this project. It has already improved my life.

jfedor2, 2024-04-02:
I realize the hub situation is not optimal. I hope to do something about it in the future.

If you've ordered boards from JLCPCB before, we do now have a custom board with a hub built in. So that's at least one configuration that's sort of guaranteed to work (though not 100% - I've found that for example a SpaceMouse Compact doesn't work when plugged into this built-in hub, even though it works with a simple HID Remapper directly).

miwagner1, 2024-04-02:
[image]

I’ll be testing tomorrow at some stores after work. I found everything I needed to test out my theory.

Without the 3.0 hub, my trackball is working with the feather rp2040 with iPhone 11 assistive touch 😮.

miwagner1, 2024-04-02:
Best Buy essentials™ - 4-Port USB 2.0 Hub - Black
Model:BE-PH2A4AT SKU:6453994
[image]

Works without any issues so far. I wonder if anyone has gotten any 3.0 hubs to work or if it’s 2.0 only.

alanbork, 2024-11-04:
Has any progress on this been made? I'd really like build a tool to combine two gamepads into one, and of course if there's two gamepads it's going to have to include a hub.

jfedor2, 2024-11-04:
> Has any progress on this been made? I'd really like build a tool to combine two gamepads into one, and of course if there's two gamepads it's going to have to include a hub.

Just find a hub that works or use one of our custom boards.

mrbeak, 2025-07-10:
I ordered 4 different usb 2 hubs. I will let you guys know which ones work and which ones don't.

Also: the second model from: https://github.com/jfedor2/hid-remapper/issues/78#issuecomment-1664502389 can still be bougth in Germany.

mrbeak, 2025-07-20:
so from 5 devices one is working as a hub with hid remapper. Most cheap hubs don't tell which chip they use so i think this info should be somewhere more prominent then here