jfedor2/hid-remapper #81: Question: Daisy chain vs Side by side dual hid-remapper?

dglaude, 2023-07-08:
We now have unique name thanks to #72.
If I put two remapper side by side on two separate USB port, I can distinguish them.

[image]
Feature request: Maybe we could rename each device with a nick name to make them meaningful.

Then I tried to daisy chain two remapper one behind the other, and in that case, the user interface only see the first(?) one in the chain.
I was wondering if there would be a way to make them all contrôlable (if not by default, with a flag).

PS: I can work around that by setting them up separately and then doing the daisy chain.

jfedor2, 2023-07-08:
A configurable name has been suggested before, might add it down the line.

Adding a way to configure a daisy-chained HID Remapper - I don't see it, it's a bit niche use case to be honest.

itsnoteasy, 2023-07-12:
There's also the issue of voltage drop with in series devices.