jfedor2/hid-remapper #83: Unresponsive after shutdown startup, needing replug

BangDroid, 2023-08-04:
Single Pico variant, latest firmware. Connected to Elecom Huge.
After shutting down Windows (10 21H2) and restarting the next day, no input was detected, neither the indicator led on the Pico was lighting. Unplugging the the Pico (with Huge still connected) and re-plugging resumed correct function. Pico is connected directly to computer rear USB not a hub.

jfedor2, 2023-08-04:
Does this happen every time?

itsnoteasy, 2023-08-05:
I have the same issue with a single pico variant plugged to the computer rear usb and a jfedor trackball. i can unplug the trackball from the pico and it wakes up. it happens every boot, but using older remapper.uf2 firmware only was happening once every 10 boots.

I made a previous post requesting a watchdog timer which i closed https://github.com/jfedor2/hid-remapper/issues/69

BangDroid, 2023-08-06:
Appears to be happening every time. I tested multiple times over the past few days and while connected to the remapper the Huge always becomes unresponsive, but without the remapper the Huge behaves as expected.

Also as itsnoteasy mentioned, unplugging the Huge from the Pico is sufficient to resume normal operation.

jfedor2, 2023-08-06:
Thanks for the reports. I need to try and reproduce this.

itsnoteasy, 2023-08-06:
I have done some more testing, and it appears that if i reboot with shutdown -r now, then the trackball does not become unresponsive. it only becomes unresponsive after a cold boot. While unresponsive, there are no messages within the monitor tab but i can open the device, and load all the settings from it.

BangDroid, 2023-08-07:
Just sharing more information and experience.
Logging in today with the Huge connected to Pico it was unresponsive on login screen, after logging in and waiting a few moments, I noticed the DPI indicator on the Huge flash and the Huge woke up and was fully responsive. The wait was maybe 30 seconds.
I'll admit I was probably impatient the previous times, standard HID behavior is usually immediate, but I suspect this is what's happening. For whatever reason the Huge is slow to wake when connected to Pico.
I may not have created this issue knowing this, it's certainly acceptable to me to have a delay, compared the having to un/replug. I will leave this open for now as maybe it's an indicator of something undesired.

itsnoteasy, 2023-08-27:
I stopped using the hid-remapper because i detested replugging, but it turned out that my trackball needed to be replugged, so i must retract my bug report, as it could have been the trackball causing the issue. my gut feeling is that both jfedor devices, the jfedor trackball and the remapper have the same bug, but i haven't used an oem mouse enough to compare with. 
my current workaround is to solder a normally closed microswitch in series in the 5v usb wire. when it locks up after boot, i click the switch.

LeoKesler, 2023-09-10:
I think I have almost the same issue, but with Elecom Huge Wireless.

I was using an older version of your firmware. After updated to latest version, the Pico is not working after a cold boot. I need to unplug and plug again the start working.

If I flash the older version, it is  back to working flawless.

1. Update to latest firmware. It start to work, including reboots.
2. At night, shutdown the computer.
3. Next day, Pico is not working anymore. 
4. Unplug Pico  and plug again and it work again.
5. Flash older firmware and it work again after cold boot at morning.

Older firmware compiled by me: 1a4eec1 .  Pi Pico single.  Windows 10.

itsnoteasy, 2023-09-10:
I have also tried this older firmware linked by leokesler. It has fixed my cold boot replug issue, but I haven't been able to program it with the browser since the tap and hold feature is missing. i noticed that with the new firmware my cursor tended to jump 5mm about once a minute on the screen. i thought that it was dust on my sensor causing it, but it is completely absent with the older firmware.

jfedor2, 2023-09-11:
This is a good lead, I should be able to git bisect, except I don't think I'm getting the issue on my PC. Will try to look around.

itsnoteasy, 2023-09-11:
I can post you my old motherboard, core i5 4670 and 16gb ram, if it would help diagnose the issue. They won't sell due to age, so i don't want any money.

itsnoteasy, 2023-10-09:
i have updated to the latest release and the cold boot issue appears to be gone. good work jfedor. had you intentionally fixed it or was the fix by accident?

jfedor2, 2023-10-09:
I haven't done anything to fix this specifically. It's possible that something changed internally in TinyUSB and we got a behavior change when we pulled the latest version in.

jasonanovak, 2024-03-14:
So, I'm in a similar-ish boat: 
- Single Pico variant, running r2024-02-14 connected to Elecom Huge wired.
- Pico is connected to a powered USB hub, which is then connected to a Chromebook + external monitor + external keyboard
- If I let the Chromebook go to sleep overnight and then wake it using the keyboard, input from the Elecom Huge + Single Pico isn't registered until I unplug and replug the Elecom Huge from the Pico.

@itsnoteasy @LeoKesler do you know what versions of the firmware you're currently running and whether or not you have the problem with it?

itsnoteasy, 2024-03-14:
I don't use a usb hub, and i dismantled the jfedor trackball, but i upgraded to r2024-02-14 single  pico variant with the kensington expert wired. the pc boots without issue with my ryzen 5. I haven't tried sleep modes, as i have them disabled. I think there's an old linux kernel bug with sleep modes so your issue may not be hid-remapper's fault.

LeoKesler, 2024-03-20:
@jasonanovak I am not using Elecom Huge Wireless anymore, because the many issues with connect / disconect and battery issues. It was a huge mistake to buy the wireless version.

Currently, I am using the Protoarc EM03. It is wireless too, but I never got a disconect from it. Almost perfect device, except for the size ( I have big hands)  and the lack of additional button at right side, like Elecom Huge.

222Phoenix, 2024-08-07:
Same issue here with single pico and Elecom Huge wired. After the PC wakes from sleep or reboot, it needs to be unplugged and replugged. I tried several other devices but only the Elecom Huge is like this. Windows 11. Using the latest firmware, r2024-06-27.

sriharshaguthikonda, 2024-09-04:
I had the same issue. I started using a different keyboard and it has solved the issue I believe. i am doing more thorough testing to see if the issue reappears somehow. will post an update here if something happens.!

 there was another issue, in general with any of the keyboards that are not connected via remapper are connected and disconnected repeatedly , not immediate connection and disconnection but repeated connection and disconnection makes the keyboard unresponsive. idk if this is the root cause for unresponsive remapper-connected keyboard. but it would be wise for others to check if it is the same with their systems.
 
 
i have recently enabled selective usb suspend setting in power settings. not disable, but enable. idk if this has solved my issue.?! ie., more testing is needed. , will update here.

i tried to create a fork of this repo and attempted to make the remapper to renegotiate the connection after PC wakes up, but i am not a good so, it didn't work.

jfedor2, 2024-12-08:
If anyone wants to help test a hypothesis, I made a test build that you can try here that might have a fix for devices becoming unresponsive ("artifact" link at the bottom):

https://github.com/jfedor2/hid-remapper/actions/runs/12217349651

The change only applies to the single Pico variant because the hypothesis is that only that variant is affected.

itsnoteasy, 2024-12-08:
I have two single pico remappers, and one is prone to become unresponsive at random times once or more a day, the other seldom gives trouble, maybe once a month, not often enough to debug. 

Given the cost of the pcb, it's easier for me to solder in a new pico. If you would like i could try the faulty pico with the test build, but it may be wasting my time and yours chasing a hardware bug.

jfedor2, 2024-12-08:
@itsnoteasy I'm specifically referring to the issue when HID Remapper itself works, but the devices connected to it stop working after a power off, restart or suspend/wakeup and require replugging.

jfedor2, 2024-12-08:
Better yet try this build, it has a slightly different variant of the potential fix:

https://github.com/jfedor2/hid-remapper/actions/runs/12225991612

222Phoenix, 2024-12-13:
Confirmed fixed on single pico and Elecom Huge wired, thank you jfedor2!

222Phoenix, 2024-12-16:
Actually, this solved one problem and created another. Now it works right after waking from a suspend, but it will just stop working after some period of time, requiring an unplug and replug. No sure if inactivity matters, doing more testing.

jfedor2, 2025-03-10:
The latest release contains the fix from the test build above.

222Phoenix, 2025-03-11:
Thank you jfedor2! I have only tested one sleep/wake so far, but it worked perfectly. 

By the way, prior to r2025-03-10, I noticed that a workaround is to click the left button several times. That seemed to wake it up without a unplug/replug.