jfedor2/hid-remapper #84: Single Pico + Elecom Huge: Can the Huge go beyond 125 Hz at all?

DryRoastedLemon, 2023-08-04:
Hello there!

I'm using a single RP Pico with the latest firmware (r2023-07-19) with an Elecom Huge (and Windows 11). It's been fantastic overal and I can't express the gratitude enough for finally being able to get rid of the MouseAssistant software forever. However, I've run into that I can't change the polling rate on the Huge. Whatever I do it ends up reading 125 Hz (or lower). When I connect a Logitech Trackman Marble to it it does go all the way up to 1000 Hz. This leads me to think that it's simply not possible with the Huge, which is fine as that's already mentioned in the project description. I'm mostly checking in if this is indeed the case with the Huge, or that I'm either doing something wrong or if I need the Dual Pico version for this.

jfedor2, 2023-08-04:
If it's the wireless version then most likely it won't go beyond its default, because we can't affect the communication between the wireless dongle and the trackball. Dual Pico version won't help.

DryRoastedLemon, 2023-08-04:
It's the wired version, actually! I forgot to mention that. Thanks for mentioning that the Dual Pico version won't make a difference; that saves me some money, ha.

jfedor2, 2023-08-04:
Oh, interesting. I got some reports in the past that the wired version doesn't work at all with HID Remapper.

But yeah, even if it's wired, if it doesn't want to go higher then there's not much we can do.

DryRoastedLemon, 2023-08-04:
Alright, thanks! Yeah, it’s working fine except for the polling rate thing. For now this will do just fine, though I might look for a different trackball in the future. I’ll close it.

Off-topic, but do you have any reports on the ProtoArc EM03 and whether it supports a higher polling rate? It’s wireless, though, but it’s the one that I’ve been looking at lately.

closed 2023-08-04 by DryRoastedLemon

jfedor2, 2023-08-04:
I don't have one, but I've read that the EM03 works at 250Hz when using the wireless dongle. Probably can't be overridden.

DryRoastedLemon, 2023-08-04:
Alright! That's already a big improvement. Thanks :)