jfedor2/hid-remapper #85: MIDI Controls

undecideduser8, 2023-08-13:
is it possible to add midi controls? #

jfedor2, 2023-08-13:
Do you want to connect a MIDI device to the Remapper and have it work as a USB keyboard or the other way round, make it emulate a MIDI device? Neither is outside the realm of possibility, but there hasn't been any specific effort in those directions.

undecideduser8, 2023-08-13:
i would like the keyboard to emulate MIDI but if it supports both methods, even better!

cyberboy666, 2023-10-19:
seconding for midi support ! im more interested in connecting a usb midi device (like korg nanokontroller) and mapping this to things... gread project btw and so well documented :)

jfedor2, 2023-10-19:
> seconding for midi support ! im more interested in connecting a usb midi device (like korg nanokontroller) and mapping this to things... gread project btw and so well documented :)

@cyberboy666 Please try the latest release, it adds support for MIDI inputs over USB!