jfedor2/hid-remapper #86: Monitor Tab doesn't show anything

xRaska, 2023-08-16:
i saw one of the first issue that people had regarding the interactive remapping funcionality, and i saw the author last message on it saying that they added the monitor tab but i can't seems to see what i am pressing there.
Am i missing something?

jfedor2, 2023-08-16:
What device are you using? Does it work otherwise?

xRaska, 2023-08-16:
sorry for the lack of information, i'm using the Seeed Studio Xiao nRF52840 with a bluetooth numpad (an huion k20).
it does work very well apart from the monitoring tab, and i need it to remap the dial, otherwise i will need to search for the right  hex value but there's hundreds of them.
is this an issue because i'm using the bluetooth microcontroller?

jfedor2, 2023-08-16:
Also if the dial is Surface Dial-compatible then it likely uses `0x00010037` code.

jfedor2, 2023-08-16:
The codes will probably be the same over USB, but it's strange that the light doesn't blink when you operate the dial. It should blink every time it receives something, even if it doesn't know what to do with it.

jfedor2, 2023-08-16:
You can try this test build, it should have the monitor tab working in the Bluetooth versions (click on artifacts at the bottom):

https://github.com/jfedor2/hid-remapper/actions/runs/5882479664

Though I suspect that the dial might still not show up if the LED isn't blinking.

xRaska, 2023-08-16:
i tested it and it works wonderfully! and i figured the problem about the dial. 

Somehow huion made it so that the dial only works with the official software running, I suppose the internal firmware detect if it's connected directly or not. i came to that conclusion by quickly swap between the bluetooth connection and the XIAO and i discovered that it works for a couple of seconds, after that it will stop working. 

I have no idea why huion decided to do that, it seems very stupid so i'll search for another bluetooth numpad/macropad.

thank again for the very quick solution to the monitoring!

xRaska, 2023-08-16:
actually i checked again, the codes that the monitor gives me doesn't seems to match with the ones that i have in input (https://imgur.com/a/3ddlohf) is that a problem on my part?

jfedor2, 2023-08-17:
Not sure if I understand the question. The hex codes at the bottom are only for inputs that are not listed above. If a code is known (has a name in the UI) then it won't be listed again in the extra section at the bottom. Is that what you meant? The monitor tab gives you both the hex code and the name (if it knows it).

xRaska, 2023-08-17:
that's what i meant yes, i just wanted to be sure, thanks a lot again, i promise this was the last question lmao.