jfedor2/hid-remapper #87: Not working with any mouse after the new update.

chalermla24, 2023-08-16:
I am using the dual version. After flashing the new firmware the mouse stopped sending any input to the PC. The mouse does still light up and the board is showing signal when moving/clicking the mouse. No idea what cause this. Rolling back to the previous firmware fixed the problem.

jfedor2, 2023-08-16:
Did you also flash the B side when upgrading? With this release you have to flash both (always a good idea).

chalermla24, 2023-08-16:
Yes,  after flashing the A side I flash the B side using the "Flash B Side" button in the website.

chalermla24, 2023-08-16:
Weird. After flashing using the BOOTSEL method instead of from the website the device now works perfectly. Should this issue be closed?

jfedor2, 2023-08-16:
"Flash B side" button works only if you're using the custom boards or if you connected the debug pins between the Picos (SWD/SWDCLK).

chalermla24, 2023-08-16:
Ah yes. That's explained the problem. I wonder why it works fine until the lastest version.

closed 2023-08-16 by chalermla24

jfedor2, 2023-08-16:
B side rarely changes so usually you can get away with not updating it.