jfedor2/hid-remapper #89: BIOS, bootloader and re-plug

philinp, 2023-09-01:
My setup:
HP Spectre x360 <--(direct USB connection, no hub)--> Adafruit Feather RP2040 with USB Type A Host (latest firmware r2023-07-19) <---> Apple Magic Keyboard 2.

1. After hard reboot or power on the laptop enters BIOS, keyboard doesn't work in BIOS even if it's re-plugged, however red led indicator is blinking on the Adafruit board while typing.
2. Then laptop enters my bootloader and asks for the password. Keyboard still doesn't work. Red led indicator is blinking while typing. Re-plug doesn't help. Re-plugging doesn't help.
3. After Windows 11 loaded, keyboard needs to be re-plugged to make it work.

If it has to be debugged in order to fix it, how do I do it?

jfedor2, 2023-09-01:
1 and 2 might be just because HID Remapper doesn't support boot protocol mode. Not sure about 3.

itsnoteasy, 2023-09-01:
generally if you want to debug software or hardware without understanding the issue, you start by breaking the whole into two parts, then testing each half separately for the bug. halves are the fastest method to find a bug. so you find that the bug is confined to one half, so then you mark the other half as good, and you divide the bugged half again in two.
This method seems slow, but it only takes about 30 dividing steps to isolate the bug.