jfedor2/hid-remapper #90: Adafruit Feather RP2040 with USB Type A Host - GPIO pins

ghangaskan, 2023-09-01:
I would like to add typewriter sound effects to the hid-remapper on RP 2040.
Like this: [Hackaday article about KB with integrated solenoid](https://hackaday.com/2023/07/19/solenoid-keyboard-sounds-very-much-like-a-typewriter/)
So I bought an [Adafruit Feather USB Host](https://learn.adafruit.com/adafruit-feather-rp2040-with-usb-type-a-host/overview)
And a [Relay Featherwing](https://learn.adafruit.com/mini-relay-featherwings)

It looks like I could use [GPIO 24 or 25](https://learn.adafruit.com/adafruit-feather-rp2040-with-usb-type-a-host/pinouts) to momentarily trigger the relay

I would like to trigger the relay on keydown event of any keys (I might want get fancy later). Can you recommend where I should place such a call?

ghangaskan, 2023-09-02:
[I took a whack at it](https://github.com/ghangaskan/hid-remapper)
[Writeup](https://hackaday.io/project/192447-keybard)

I vamped on the LED activity function calls to make relay clacks related to keypresses.
I used GPIO 10 instead of 25 and 25.

I may come back later and leverage the expressions interpreter to set variables? Not sure.
I may also get excited to capture keydowns only, and maybe just for most keys, but exclude CR, LF, PgDn, Enter, (40 keypresses since the last CR) etc. If I could map thos to different pins, I would have everything I'm hoping for...

ghangaskan, 2023-09-03:
Ok, so, I made it through proof of concept, and in digging into the code I noticed you have the GPIOs listed as available inputs and outputs for mapping. However, the code only reads from GPIO.

I would like to set GPIO bitmask to in/out/off on the settings tab. Then have GPIO init configure for input and output. Perhaps set a global output delay? Not sure about that.

I will take a first pass at it, but c++ is not my native language, so even if I get something going, I would appreciate help getting my concepts right.

jfedor2, 2023-09-03:
Yeah, I considered adding GPIO outputs as well, for LEDs for example, to visualize layer state or as an extra CapsLock, but never got around to it.

ghangaskan, 2023-09-06:
I failed trying to get real native with it.
Tried to use bitmask to set which IO ports were read and which were write, and in the end just made a mess of my code. I may try again later. If you have some time I would love some recommendations on how to be successful and cross-platform compliant

jfedor2, 2024-01-02:
I forgot to mention it here, but GPIO outputs are now supported!