jfedor2/hid-remapper #91: Using the tap flag results in multiple outputs

DryRoastedLemon, 2023-09-03:
I use a ProtoArc EM03 (wireless) and I like using the back button in a tap-hold configuration. When I tap the button it'll function like a regular back key, but if I hold it I can scroll by moving the trackball. However, I noticed that when I assign the Tap flag to the regular Back output (so Back -> Back [Tap]) it'll actually often get sent multiple times! This even happens if I only configure the tap functionality and don't configure anything for the held condition. I assign toggle to mute to this button in Discord and I've noticed that it'll sometimes very quickly mute and unmute me again. This happens on my wired Logitech Trackman Marble too, so it's probably not the wireless connection. Is there something wrong in my configuration, or is this a bug?

jfedor2, 2023-09-03:
Can you post an example configuration in which it happens? I will try and reproduce.

DryRoastedLemon, 2023-09-03:
Sure! It essentially comes down to this configuration:

[image]

Here's my complete configuration, which includes a bind which remaps the forward button to a middle mouse button click.

`
{
    "version": 8,
    "unmapped_passthrough_layers": [
        0,
        1,
        2,
        3
    ],
    "partial_scroll_timeout": 1000000,
    "tap_hold_threshold": 200000,
    "gpio_debounce_time_ms": 5,
    "interval_override": 0,
    "mappings": [
        {
            "target_usage": "0x00090003",
            "source_usage": "0x00090005",
            "scaling": 1000,
            "layers": [
                0
            ],
            "sticky": false,
            "tap": false,
            "hold": false
        },
        {
            "target_usage": "0x00090004",
            "source_usage": "0x00090004",
            "scaling": 1000,
            "layers": [
                0
            ],
            "sticky": false,
            "tap": true,
            "hold": false
        }
    ],
    "macros": [
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        [],
        []
    ],
    "expressions": [
        "",
        "",
        "",
        "",
        "",
        "",
        "",
        ""
    ]
}
`

jfedor2, 2023-09-03:
I know you've said you're seeing this on two separate devices, but are you sure it's not a bouncing issue on the switch? Can you try the same config on other buttons? Because I'm not seeing it in my tests.

DryRoastedLemon, 2023-09-04:
I’ve also tried it on the forward and right mouse buttons with the same results. Very strange! I’ll fiddle around to see if it could be a software problem, as I’m also running Steermouse on MacOS and X-Mouse Button Control on Windows.

DryRoastedLemon, 2023-09-04:
I've done a little more fiddling around, and I think I've made a little bit of progress on the issue. I've found that this only happens to me on MacOS! I think my issues on Windows are at least partially down to user error - though I have found in some cases that sometimes mouseclicks don't seem to register. Will do some more research on that later.

Anyway, the issue I'm describing is happening on Mac, and the issue is kind of interesting. Without using the HID Remapper device intercepting the inputs there's no issue whatsoever, and Discord sees the buttons as regular single presses, i.e. "MOUSE4", "MOUSE5", etc. When I HID Remapper _with no binds set_ Discord suddenly reads double presses, namely "MOUSE4 + MOUSE4" and "MOUSE5 + MOUSE5". I think this results in the strange behavior I'm getting! Is there anything I can do?