jfedor2/hid-remapper #92: PS4 Gamepad HID Controller - no output in Monitor tab - Single Pico board

HVR88, 2023-09-13:
Chrome browser in Mac OS 13.5.1

I'm using a single Pico board that's working as expected with USB mouse connected (axis, all buttons, wheel), however, when i connect an 8BitDo SN30 Pro gamepad in PS4 mode (Mac OS/HID compatibility) I get no output from the monitor tab.

I've verified the controller is working and recognized as PS4 pad using gamepad-tool which uses SDL2.  VID/PID = 1356/1476 usage 1/5

Should this be working with the single Pico board or is it something expected to only work with the double?

jfedor2, 2023-09-13:
Hard to say, I don't have one to test. There's no guarantee it will work with the dual Pico version, but it might. Does the LED on the Pico blink when you press buttons on the controller?

HVR88, 2023-09-13:
It doesn't blink when the game controller is connected and any buttons are pressed.  It does blink with every movement/button of the mouse I mentioned above.

I have a couple more Picos here so I can try the dual setup later in the week.

Do you know of any gamepads that have been tested as working?

jfedor2, 2023-09-13:
The Stadia controller works.

HVR88, 2023-10-10:
Thanks for the feedback, I'll give the board another try with some first-party PS controllers I have here.

HVR88, 2023-10-11:
It seems to be working now with the latest firmware. PS4 controller AND the 8BitDo in both Android mode and PS/Mac mode.

I've also tested XBOX One controller and all buttons/axis work there too. 

Oddly, the 8BitDo controller in XBOX mode doesn't work. Maybe just need to add the PID

Real XBOX One Controller:

  Product ID:	0x0b12
  Vendor ID:	0x045e  (Microsoft Corporation)
  Version:	5.0b
  Serial Number:	3039373133383239323936303432
  Manufacturer:	Microsoft

8BitDo in XBOX Mode:

  Product ID:	0x028e
  Vendor ID:	0x045e  (Microsoft Corporation)
  Version:	1.14
  Serial Number:	157F8F8
  Manufacturer:	8Bitdo SF30 Pro

jfedor2, 2023-10-11:
Thanks for testing!

I think the 8BitDo controller pretends to be an Xbox 360 pad. Only Xbox One/Series controllers are supported, at least for now.

HVR88, 2023-10-11:
I updated with more details after testing both XB One and XB Series controllers. :)

The 8BitDo in "Android" mode is using DInput and in XBOX mode uses XInput - but honestly, I don't know what other parameters it has/lacks to make it look like X360 vs. XBOne/Series.

jfedor2, 2023-10-11:
I think the wireless adapter is a different beast, it doesn't just show up as a wired controller. So adding support for that would be a separate undertaking.

Thanks for the continued testing!

HVR88, 2023-10-12:
I'm going to see about getting the BT board so I can test that at the same time. I also have a few more USB controllers kicking around I can run through this when I have some time.

jfedor2, 2023-10-24:
> Hello, may I ask which controllers you have tested that can successfully pair with the nrf52840's HID-Remapper? I have a Stadia and a DualShock 4 controller, and neither of them can successfully pair with the HID-Remapper (using the Seeed XIAO BLE nrf52840, the blue LED stays on. I have tried both shorting GND-P0 and using the WEBUI button to enter pairing mode, but I can only pair my Logitech MX Master2s mouse, not the controllers).

DS4 uses Bluetooth Classic so it won't work.

Stadia controller uses Bluetooth LE, but for some reason it doesn't show up in scans, I'm not sure why.

HVR88, 2023-10-24:
So far I've only used the XBOX One (X) controller.